prusa3d/PrusaSlicer #2838: UI elements order logic VS actual position - need some adjustement

TeldonComputer, 2019-08-29:
### Version
2.0.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
windows 10 Pro 1903
build 18362.10015 (inside program)

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3SMMU2 
firmware prusa3d_fw_3_7_2_MK3S_1_0_6_MMU2S

### Behavior
the right panel present elements in wrong order.

[image]

Changing Printer affect others menu', so it would be more logic to put the "printer" selection on top.

_Is this a new feature request?_
No, only a (minor) UI modification.

Thank you all for all the work!

jokohoko, 2019-08-30:
I support this request since changing an option from the third option (Printer) now changes the options in the top option (Print settings).
It will suck for some days since it has became automatic by now, but it would make it a lot more logical.

bubnikv, 2019-08-30:
This has already been discussed, management decided to keep it the way it is. Closing.

closed 2019-08-30 by bubnikv