prusa3d/PrusaSlicer #2839: Slicer slice some holes instead of object

ondratu, 2019-08-29:
### Version
Version:   2.1.0-beta2+linux64
Build:     PrusaSlicer-2.1.0-beta2+linux64-201908250731

### Operating system type + version
Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.2.0-2-amd64 x86_64
Total RAM size [MB]: 16,574MB
OpenGL installation
GL version:   3.0 Mesa 19.1.4
Vendor:       Intel Open Source Technology Center
Renderer:     Mesa DRI Intel(R) UHD Graphics 630 (Coffeelake 3x8 GT2) 
GLSL version: 1.30

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3/S

### Behavior
Import STL model -> Slice
[image]

Object:
[bb-a90-3x4.zip](https://github.com/prusa3d/PrusaSlicer/files/3555058/bb-a90-3x4.zip)
Source:

bubnikv, 2019-08-29:
It slices correctly if fixed with Netfabb.

ondratu, 2019-08-29:
STL from last OpenScad (OpenSCAD version 2019.05), works right. Main difference between the STLs, are less precision in numbers. Resp. new STL does not have any too long numbers in "e-" format.

git-prusaslicer, 2019-08-30:
I seem to have the same issue.  Works fine in 2.0 but in 2.1 a 2mm wall is missing.  I would attach the scad and stl files but it seems they are not supported as attachments here.  Generated from openscad.

[image]
[image]

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo