prusa3d/PrusaSlicer #2842: Some UI improvements: Filaments list

jokohoko, 2019-08-30:
### Version
_V2.1_

### Feature request:
It would be great to have an option of selecting commonly used filaments so that all other filaments get either removed from the list or showed lower in the list then the generally used ones.
My proposal would be to show the list of filaments in the settings with a checkbox in front. Users can then select or deselect filaments to make their actively used list.

For example having multiple PLA's like Prusament, Generic and Prusa PLA makes it difficult to explain to new users. Especially since i know the differences are so small that it would not even matter.

bubnikv, 2019-08-30:
@vojtechkral is currently working on it.

rtyr, 2021-02-03:
Implemented since 2.2.0-alpha1. Closing.

closed 2021-02-03 by rtyr