prusa3d/PrusaSlicer #2844: 2.1.0-beta2 Linux AppImage - OpenGL detection fails

BerndJM, 2019-08-30:
While all version before (including the alpha1) where fine with my OpenGL-Version
[image]

in the beta2 the detection fails ...

[image]
[image]

bubnikv, 2019-08-30:
Are you saying that on that particular system, the 2.1.0-alpha1 still runs, while the 2.1.0-beta2 fails to detect OpenGL?

bubnikv, 2019-08-30:
I see the cause now. On Linux the OpenGL gets first available at the first render event, but the check is done much earlier.

Fixed with b305a6b37eced7eddba430db7e5991dd6b9c45b5. Thanks for the report.

closed 2019-08-30 by bubnikv

BerndJM, 2019-08-30:
Just an additional info:
I had at the same time the problem with the upcoming sanity-check ( #2845 ).
Seems like the failing OpenGL detection is "triggered" by this upcoming sanity-check info-window.
As soon as I get rid of the sanity-check message even the OpenGL detection works fine ...

bubnikv, 2019-08-31:
We are going to release the beta3 now. We will look into the "sanity check" for the final release.