prusa3d/PrusaSlicer #2845: 2.1.0-beta2 Linux AppImage - Annoying parameter check on startup

BerndJM, 2019-08-30:
Each time the program starts, a Parameter validation hint appears.
I have not the slightest idea where these 1,2% should be?!?

[image]

Also, it does not make a difference if I click "Yes" or "No", the note appears again every time.

Jebtrix, 2019-08-30:
Your on linux so just search in terminal:

`grep --color -R '1.2' ~/.PrusaSlicer-beta/`

Result will look something like:

`/home/<username>/.PrusaSlicer-beta/print/0.15mm OPTIMAL  Demo.ini:infill_overlap = 1.2`

BerndJM, 2019-08-30:
Thanks a  lot for this search. I managed to find it in the support material XY seperation.

Beside the question if this value makes sense here, there is the fact that this "autocorrection" only works temporarily. If I choose "Yes" to change the value to % the value gets changed, but this will be reverted when I change the profile or close the Slicer.

On the other hand is the question about the sense of the check and correction in this case, if it provides an absolute nonsense value as replacement?

Jebtrix, 2019-08-30:
Once you choose "Yes" do a sanity check the value has changed in the UI. Then make sure your save the profile otherwise the change is temporary. 
_Take note if you save a Project to a file. When you open that file it's using the embedded profiles._

Jebtrix, 2019-08-30:
As for why its complaining about 1.2mm is that they have added sanity checks to values. I personally don't see a problem with 1.2mm X-Y separation. But a value getting that large you have to be more careful of the result. Take this cup for example at that value of 1.2mm :

Using Support on Build Plate Only setting:
[image]

This isn't really bad. Maybe you just want the base of the handle supported but steer clear of the rest without extra work of support enforce/blocker modifiers.

Using Support on Everywhere setting:
[image]

Well now we have a problem. That inner handle support material isn't possible to physically print. That value sanity check is to try to prevent this.

If we use a more _normal_ value of 0.15mm we get a more sane and printable result:
[image]

BerndJM, 2019-08-30:
Thanks again for this examples. Well I see the difficulties appearing with this value, even if it works in my use cases perfect ;-)
As this parameter expressed in % is calculated over the external_perimeter_width I can set it to 300% and both will be happy, the sanity check and me ;-)
Nevertheless the - by the sanity check - suggested 1.2% are a bit "funny" (don't know a better word).

Jebtrix, 2019-08-30:
> Well I see the difficulties appearing with this value, even if it works in my use cases perfect ;-)

I completely understand. Choosing values can be a small form of art sometimes. The devs are just trying to catch human errors but the use cases are so vast. I've done it myself typing in 5 when I meant 5%. That is where the simple suggestion of the percentage counterpart comes in.

Jebtrix, 2019-08-30:
You should close this issue unless saving the profile didn't get rid of the warning message on startup. You can change it to a Feature Request if you want to ask for that parameter warning to actually display the parameter name its referencing.

closed 2019-08-30 by BerndJM