prusa3d/PrusaSlicer #2846: Try to delay very short perimeters after bigger ones in first layer

guestisp, 2019-08-30:
### Version
all

Look at the attached gcode. PrusaSlicer starts to print some very very very small parts (mostly, a single dot) and immediatly after start to print a much bigger perimeter all around it.

Single dots or very small perimeters are very hard to stick properly on a plate, even with hairspray.
If there are a much bigger perimeter to print immediatly after, PrusaSlicer should print the bigger perimeter and then the small one that would stick to the bigger perimeter.

As PrusaSlicer is already able to detect what is a small perimeter, this small perimeter should be printer, as much as possible, after everything else, on first layer.

Or, a feature to print external perimeters first on first layer only (you can't have any overhang on first layer, so there is no advantage in printing the inner perimeter at first). The external perimeter is always bigger than the inner one.

On this specific case, setting external perimeter first would solve the issue.

could be related to #2165 

[top-plate-R.zip](https://github.com/prusa3d/PrusaSlicer/files/3558993/top-plate-R.zip)

Jebtrix, 2019-08-30:
Well a particular model that may work.

Jebtrix, 2019-08-30:
There is already external perimeters first option that you can use a modifier to use it on just first layer.

guestisp, 2019-08-30:
I wasn't aware of that as modifier, i'll check, thank you

Jebtrix, 2019-08-30:
[image]
[image]

guestisp, 2019-08-30:
But how to set a modifier only on a single layer?

Jebtrix, 2019-08-30:
If your using the beta you can use either layer range or a slab modifier. For a slab modifier you set it's height to the first layer height. For the layer range just make its Max Z and Layer Height = your first layer height.

Jebtrix, 2019-08-30:
You can Close this issue then :)

Jebtrix, 2019-08-30:
Just a fyi, you can also use Infill Before Perimeters _sometimes_ to achieve the same goal. Just make sure your in Expert Mode to see these options.

guestisp, 2019-09-01:
Trying this feature right now.

But, any particular advantage in doing the inner perimeter first at the first layer ? I don't see any advantage, only drawbacks.

IMHO, the first layer should be always printed with external perimeters first.

albert2004, 2019-09-04:
@guestisp I don't agree. Take for example flooding nozzle. If you always start first layer from external perimeter then you will always have underextrusion at beginning. When internal perimiters are extruder first, then flow will be correct when it will come to external one. Also possible blobs will hide inside of object and will not be visible.
This is just an example, where your suggestion is not right.

guestisp, 2019-09-09:
@albert2004 ok, so let's keep the default as it is currently

But an option like "external perimeters starting at layer #" or "internal perimeters starting at layer #" would be great. Actually i have to use the layer customizer, every time, and the value Is not preserverd in the generated gcode

So, adding an option that could be saved in gcode and in profile, would be great and much quicker to use

closed 2026-09-29 by prusaslicer-bot[bot]