prusa3d/PrusaSlicer #2847: New functionality - quantity or factor of extruded filament in the supports

hardmax, 2019-08-30:
Hi,

On some occasion it happened to me to change the nozzle (0.8mm) and print an object that was made for a 0.4mm nozzle. 
The result was an impression very similar to the original, but with a very fine density, very soft and easily breakable. 
I suppose that a lower amount of fillament when print, do this,. 

This results would be perfect for supports, low material, easily to remove and very easy to implement......

Regards

bubnikv, 2019-08-30:
Extrusion width settings for supports is already there. We use it at Prusa
Research to set the extrusion width of supports to 0.3mm for a 0.4mm nozzle
diameter.

hardmax, 2019-08-30:
Sorry,

But I Ussing PrusaSlicer 2.0.0 and I don´t view that. Maybe can you help where is It?

Jebtrix, 2019-08-30:
Make sure your on Configuration > Mode > Advanced OR Export mode

[image]

rtyr, 2021-02-03:
Answered. Closing.

closed 2021-02-03 by rtyr