prusa3d/PrusaSlicer #2848: Supports in beta2 causing empty layers

JasXSL, 2019-08-31:
### Version
2.1.0-beta2+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
MMU2s

### Behavior
Adding supports creates empty layers, and a warning that the print will not work right. It seems like the supports are being printed with the wrong layer height. Using the default 0.3mm DRAFT setting, and enabling supports, allowing the program to auto-generate support settings.

[image]

#### Project File (.3MF) where problem occurs
[speaker_test.zip](https://github.com/prusa3d/PrusaSlicer/files/3561939/speaker_test.zip)

Jebtrix, 2019-08-31:
It is because your model isn't one mesh. If you split it to parts and look at layer height problem, you can see which pieces of the model are not quite detected/touching on that layer. I started from the first, selected the two tabs and adjusted -Z till it went to the next trouble spot in the layer height with the other tabs. Repeat until no errors. Similar to this issue [#2778](https://github.com/prusa3d/PrusaSlicer/issues/2778)

[speaker_test_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/3561989/speaker_test_fixed.zip)

Jebtrix, 2019-08-31:
Look at your model closer its not the same issue as issue 2778, just same result.  Your pieces are intersecting and should be slicing just fine.

[image]

JasXSL, 2019-08-31:
Yep the model slices fine without supports. It's only when supports are on that it gets that error.

Jebtrix, 2019-08-31:
If you lower Print Settings > Support Material > Overhang Threshold to 40 degrees it will slice OK. It's trying to fit support material in the tiny tab overhang when it has no space to do so.

JasXSL, 2019-12-21:
Been a few versions and I haven't had this issue since, so I'll close this

closed 2019-12-21 by JasXSL