prusa3d/PrusaSlicer #2850: PrusaSlicer 2.1.0-beta3+win64 debug check when saving custom filament.

TheDuggem, 2019-08-31:
Version 
2.1.0-beta3+win64

Windows 7 Home Premium x64

Prusa i3 Mk3S MMU2S

Creating a new filament profile results in a debug check.

To reproduce:
Select Original Prusa i3 MK3S MMU2S Single printer
Select Generic PLA filament
Change printing temperatures on the filament tab.
Save new profile as Hatchbox PLA
Debug check occurs.

No model loaded.

[image]

TheDuggem, 2019-08-31:
Check also occurs when loading existing custom profile(s) on program execution.

Shama71, 2019-08-31:
I get the same window when start the program, about 15 times for 32 & 64 bit.

YuSanka, 2019-09-02:
Thanks for your report. This bug was fixed and will be a part of upcoming release.

closed 2019-09-02 by YuSanka