prusa3d/PrusaSlicer #2851: Feature Request: Add Nozzle Temperature Adjustment as an option in layer editing mode

FilamentFrenzy, 2019-08-31:
Would be great to add nozzle temperature adjustment as an option in the new layer editing mode and maybe even as option in mesh modifiers too!

This would make it simple to change nozzle temp at set layer heights, applications could be something as simple as temperature towers or when using colour swap and the material swapped to requires a different temperature!

Thanks,

Tom

I-like-stuff, 2019-09-28:
Yes please add this I have been waiting forever for this to be added.

rcmaniac25, 2019-09-28:
This, and maybe other filament settings like fan speeds

cmburg, 2020-04-11:
Newer versions (like 2.2.0-rc4) have the ability to add custom gcode at layers. I just found this out today:
After import/opening the desired model switch to the preview "window" (e.g. ctrl+6). Between the preview and the settings window is an orange bar with a grey horizontal line appearing whon you hover your mouse pointer above the line. Then click left and an upwards pointing triangle with a plus-symbol to the right appears. Make a right click onto that plus-symbol and choose "add custom gcode". Then enter your desired codes (multiple seem to be possible; I just tried temp change)

rcmaniac25, 2020-04-11:
IMHO:

- Most user-friendly/easy: Slicer does all the work for you (such as saying "I want to change filament from Prusament PLA to Prusament PETG" for example)
- Intermediate: layer editing (specify temperature, fan, etc.)
- Advanced: specify the exact gcode you want

So it works, but I'm not sure it's at the level everyone wants.

closed 2022-09-13 by FilamentFrenzy