prusa3d/PrusaSlicer #2853: MK3S MMU2S single filament prints broken - printer resets

rackley096794, 2019-08-31:
### Version
2.1.0-beta3

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S 3.7.2
MMU2S 1.0.6 

### Behavior
* MMU2S single-filament prints fail before starting.  The printer never prompts for which filament to use.  It heats the bed and nozzle, performs bed levelling, then the printer does a hard reset on itself and reboots.

#### Project File (.3MF) where problem occurs
[control_case_top.zip](https://github.com/prusa3d/PrusaSlicer/files/3562779/control_case_top.zip)

vintagepc, 2019-08-31:
Check your IR sensor. This is known to happen if the FINDA reads 0 while the IR sensor reads 1.

rackley096794, 2019-09-01:
Cancel this - I had the wrong printer selected (MK3).  Apparently I imported this as a user preset as part of a config bundle and it turned into the default printer type :-/

closed 2019-09-01 by rackley096794