prusa3d/PrusaSlicer #2857: Extra overhanging perimeters are added in 2.1.0-beta3

djr7C4, 2019-09-01:
### Version
2.1.0-beta3-linux64

### Operating system type + version
Linux balerion 5.2.3-arch1-1-ARCH #1 SMP PREEMPT Fri Jul 26 08:13:47 UTC 2019 x86_64 GNU/Linux

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3
Firmware: 3.7.2

### Behavior
When slicing, unnecessary overhanging perimeters are printed in thin air for some models.  These are not part of the model but occur between different parts of the model and will result in extensive stringing if printed.  The extra perimeters can be seen in the preview after slicing. 
 Slicing was done using the following settings (no changes were made from the defaults):
Print settings: 0.20mm QUALITY MK3
Filament: Generic PLA
Printer: Original Prusa i3 MK3

A model that this occurs with is shown here: http://www.thingiverse.com/thing:53451

This appears to be a regression in 2.1.0-beta3 that was not present in beta2.

#### Project File (.3MF) where problem occurs
An example project file is available here: https://drive.google.com/file/d/1TbDZYVC5mGpcF9F3HDM6RKFJF1j4uPop/view?usp=sharing

BerndJM, 2019-09-02:
In my opinion this are travel moves. Seems like there is something broken in the beta3 with displaying them.
In earlier versions you have to click the checkbox to show them - in beta3 they are shown directly without interaction needed. Additional it seems clicking the travel checkbox displays the retractions?!?

djr7C4, 2019-09-02:
@BerndJM They may well be travel moves.  However, they are shown in blue which is the color for overhanging perimeters.

bubnikv, 2019-09-02:
I reworked the G-code preview to work with lower memory graphics cards.
Maybe I broke something.

bubnikv, 2019-09-02:
Thanks for the report. It has been fixed with https://github.com/prusa3d/PrusaSlicer/commit/206dbb81c7b54a32bf2a06c3812c070072b33e85
A PrusaSlicer-2.1.0 release candidate will be released soon.
Closing

closed 2019-09-02 by bubnikv