prusa3d/PrusaSlicer #2860: Interface Layers are generatet in the air

AndyWo83, 2019-09-02:
### Version
PrusaSlicer-2.1.0-beta3+win64-201908301447

### Operating system type + version
Win7 64bit

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
Interface Layers are generatet in the air.
I dont't know why...
[image]
[lightsaber-handle_rebuilt.zip](https://github.com/prusa3d/PrusaSlicer/files/3566936/lightsaber-handle_rebuilt.zip)

Jebtrix, 2019-09-03:
Yeah I get the same issue.  I also noticed from another issue [(#2845)](https://github.com/prusa3d/PrusaSlicer/issues/2845) that the right side of a model with supports seems to be problematic. Right side tends to be thinner compared to the left side with generated supports on a symmetrical model.

[image]

###### _Arch Linux -  build 2.1.0-beta3-20-g9e7367457_

lukasmatena, 2019-09-03:
This issue seems to appear in all versions back to Slic3rPE 1.41.3 and possibly before.
I'm not saying this shouldn't be fixed, just telling others not to look for this bug in recent commits.

JustinFarris, 2019-09-06:
Anybody know of a link to an upstream bug for this? I can't find one. This has started popping up a lot more often, and it's a bit tedious sometimes to have to add a strategic support blocker to prevent what might be a bug.

Thing is, it very rarely causes any real problems, but when it does it... well, does.

Jebtrix, 2019-09-06:
Slic3r's support generation result is completely different than PrusaSlicer. I don't think upstream applies.

JustinFarris, 2019-09-06:
Ah, I see. Well, if this gets a "volunteer needed" label I'd be happy to take a very green crack at it, but it seems like something that a dev with even slightly more smarts than I have could take care of quickly.

bubnikv, 2021-02-22:
Cannot reproduce with PrusaSlicer 2.3.0.
Closing.

closed 2021-02-22 by bubnikv