prusa3d/PrusaSlicer #2864: How to delete filament profiles with lock icon?

delious87, 2019-09-04:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-09-05:
There is multiple ways. I'll give you simplest one with minimum change. **Disclaimer: Touching vendor file isn't recommended.** If you let PrusaSlicer update profile presets it will automatically overwrite this. Easiest way to clear all vendor filaments from sight (but not deleted) is to edit PrusaResearch.ini in C:\users\username\appdata\prusa-xxxxx\vendor. 

Find
```
[filament:*common*]
compatible_printers=
```

Change to 
```
[filament:*common*]
compatible_printers=hidden
```

To see these again in the UI, maybe to check a value, save a Printer profile named **hidden**. They will all show up once you change to that Printer profile (may need to restart program once). The **hidden** isn't special word, you can call it anything unique. 

If you want to bring some out from hiding, change to the "hidden" named Printer profile, go to Filament Settings which will be the full preset list, select Dependencies > Compatible printers > either click All checkbox or hit Set and select printers manually. Then you will have to save as a different name because you can't save over preset. It is effectively unhidden.

delious87, 2019-09-05:
> There is multiple ways. I'll give you simplest one with minimum change. **Disclaimer: Touching vendor file isn't recommended.** If you let PrusaSlicer update profile presets it will automatically [...]

It is not working.

Jebtrix, 2019-09-05:
I use this workaround on linux and window builds. I just applied it to the PrusaSlicer 2.1.0-rc build. It works fine. I did state the path wrong I meant:

PrusaSlicer 2.1.0-rc path:
C:\Users\username\AppData\Roaming\PrusaSlicer\vendor

PrusaSlicer-beta path:
C:\Users\username\AppData\Roaming\PrusaSlicer-beta\vendor

Make sure the latest PrusaSlicer preset update wasn't applied over it. Also make sure whatever value you use as compatible_printers=,  ie hidden, create the new Printer profile in the UI with the EXACT same name (case sensitive). If you didn't, delete the profile and save a new one with the correct name. On windows if you change printer profile name from say: Hidden to hidden it will not actually change the name. Once you restart it will be back to Hidden.

bubnikv, 2019-09-06:
@vojtechkral is working on extension of the installation wizard for system
filaments.

rtyr, 2020-05-22:
Implemented in 2.2.0. Closing.

closed 2020-05-22 by rtyr