prusa3d/PrusaSlicer #2865: Possible to create shape/object/part/your mom without main object

Zemistr, 2019-09-04:
Let's say I just opened the slicer and I would like to print a few test objects. 
But I don't have them. 😞 
Currently, I can add an object and add some shapes as parts of this object and delete the main object.
Buuuuut... What if instead of the basic object, I can just right click on the right sidebar and hit some "Create shape/object/part/your mom" button? 🤔😮

Sembazuru, 2019-09-18:
I would find this useful for setting up tests. Usually I want just a simple rectangular prism shape for me to scale in each axis to my heart's content. Currently I just use a 1cm cube STL, but it would be nice to not have to rely on me remembering where I put that sample cube or having to create a new cube in a different piece of software.

+1 for adding simple primitives to the print area without needing to load an external object first.

Zemistr, 2019-09-21:
@YuSanka Wait... You really did it? 😮 
You are amazing! Looking forward to testing it! ❤️

FidelCapo, 2019-09-26:
Thank you for feature request. This feature will be part of next release.
Closing

closed 2019-09-26 by FidelCapo