prusa3d/PrusaSlicer #2866: Possible problem with src/CMakeLists.txt

ggallant571, 2019-09-05:
#1865 # Version
_Version of PrusaSlicer used goes here_2.1.6-rc

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Linux Fedora 30
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_The variable wxWidgets_FOUND appears to be uninitialized prior to use on line 46. Commenting out the "if" allows the build to proceed. 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-09-05:
That variable is set by the FindwxWidgets.cmake module, usually found here: /usr/share/cmake-3.15/Modules/

You should be able to run `wx-config --list`. If your wxwidgets installation isn't right I'm surprised it finished a compile.

vojtechkral, 2019-09-05:
@ggallant571 Have you had a look at the [How to build on Linux](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) doc? If you're looking to build against wxWidgets 3.0, you need `-DSLIC3R_WX_STABLE=1` . Otherwise please provide more information about your setup.

ggallant571, 2019-09-05:
Downloaded prusa slicer master from github as a zip.
Installed xwWidgets via Fedora dnf. It provided wx-3.0 in the typical 
system locations.

I then created an empty build directory and from there executed:

cmake .. -DDSLIC3R_WX_STABLE=1

This failed to complete the cmake process with error messages indicating 
wxWidgets not available.

Downloaded prusa slicer release 2.1.0-rc and repeated the build process. 
Same results.

Downloaded, built, and installed wxWidgets from source. It installed in 
/user/local/include/wx-3.1 and /usr/local/lib64/xw

Same cmake results.

Commented out the CMakeLists.txt generating the failure message and got 
further. Needed the cereal package. Installed cereal-devel via dnf and 
the cmake completes without error on the master but not on the release. 
Make failed after a large number of compiles again complainong about a 
missing xwWidgets include file. The offending line (I did not save) was 
in the form #include "wx/xxx.h" indicating to me that the offending code 
was in xwWidgets.

Added symbolic links from /usr/local/include/wx-3.1 to /usr/include.

Building prusa slicer master succeeds!!! Version 2.2.0-rc still fails.

I suspect that wxWindows 3.1 must be installed in the standard system 
locations.

vojtechkral, 2019-09-05:
> cmake .. -DDSLIC3R_WX_STABLE=1
>
> This failed to complete the cmake process with error messages indicating
wxWidgets not available.

Did you install the devel variant of the wxwidgets package? Also in the quoted line above there's one `D` too many in `-DDSLIC3R_WX_STABLE=1` ... it's supposed to just be `-DSLIC3R_WX_STABLE=1`.

> I suspect that wxWindows 3.1 must be installed in the standard system
locations.

The CMake files look for wxWidgets 3.1 in standard system locations by default. With `-DSLIC3R_WX_STABLE=1` they look for wxWidgets 3.0 in standard system locations. Installing only runtime libraries variant of the package won't do, the build process also needs header files.

closed 2019-09-05 by vojtechkral

ggallant571, 2019-09-05:
The "-DDSLIC3R_WX_STABLE" is my typeo. Issuing:

 ??cmake .. -DSLIC3R_WX_STABLE=1
 ??make

cmake completes but make still fails with:

CMake Error at 
/usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
 ?? Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
 ?? wxWidgets_INCLUDE_DIRS) (Required is at least version "3.0")

ls -l /usr/lib64 returns

bash-5.0$ ls -l /usr/lib64/wx
total 12
drwxr-xr-x 3 root root 4096 Aug 14 19:17 3.0
drwxr-xr-x 2 root root 4096 Aug 14 19:17 config
drwxr-xr-x 3 root root 4096 Aug 14 19:17 include

ggallant571, 2019-09-06:
I think the real problem is cmake is failing to detect pthread_create. 
The src/CMakeList.txt line 41 is reporting it as a wxWidgets failure.

The attached C code makes no sense to me. Why use pthread_create as a 
pointer if it doesn't exist?

 From the error log:

Determining if the pthread_create exist failed with the following output:
Change Dir: 
/home/ggallant/Prusa/PrusaSlicer-master/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake cmTC_cc6a2/fast
/usr/bin/gmake -f CMakeFiles/cmTC_cc6a2.dir/build.make 
CMakeFiles/cmTC_cc6a2.dir/build
gmake[1]: Entering directory 
'/home/ggallant/Prusa/PrusaSlicer-master/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_cc6a2.dir/CheckSymbolExists.c.o
/usr/lib64/ccache/cc???? -fPIE???? -o 
CMakeFiles/cmTC_cc6a2.dir/CheckSymbolExists.c.o???? -c 
/home/ggallant/Prusa/PrusaSlicer-master/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_cc6a2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_cc6a2.dir/link.txt 
--verbose=1
/usr/lib64/ccache/cc?????????? -rdynamic 
CMakeFiles/cmTC_cc6a2.dir/CheckSymbolExists.c.o?? -o cmTC_cc6a2
/usr/bin/ld: CMakeFiles/cmTC_cc6a2.dir/CheckSymbolExists.c.o: in 
function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_cc6a2.dir/build.make:87: cmTC_cc6a2] Error 1
gmake[1]: Leaving directory 
'/home/ggallant/Prusa/PrusaSlicer-master/build/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:121: cmTC_cc6a2/fast] Error 2

File 
/home/ggallant/Prusa/PrusaSlicer-master/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
 ?? (void)argv;
#ifndef pthread_create
 ?? return ((int*)(&pthread_create))[argc];
#else
 ?? (void)argc;
 ?? return 0;
#endif
}

vojtechkral, 2019-09-06:
@ggallant571 

> I think the real problem is cmake is failing to detect pthread_create.
The src/CMakeList.txt line 41 is reporting it as a wxWidgets failure.

I don't think so, the `pthread_create` failure is noise as far as I can tell.
It probably just can't find the wxWidgets for some reason.
One reason might be GTK version mismatch. What GTK is the wx package on Fedora using?
PrusaSlicer looks for wxWidgets with gtk2 by default, so you need the appropriate wxgtk variant.
You can also try building the slicer with `-DSLIC3R_GTK=3`  added to the command line to make it look for wx with gtk3 as well, but this setup is sort of experimental and there may be visual glitches in the GUI.

I've just built slicer on my system (Arch) against wx 3.0 gtk2 and wx 3.0 gtk3 and both build succefully. So on your end it's either the gtk2/3 mismatch or some other system-specific problem making build scripts not find wxWidgets.

ggallant571, 2019-09-06:
I agree that the pthread?? is noise. I added dsplay message statements to 
the Prusa src/CMakeFile.txt an cmake FindwxWidgets,cmake.

The cmake function detects wx-config and sets wxWidget_FOUND to true, It 
the tries to find the libraries, fails, and sets wxWidget_FOUND to false;

The Prusa src/CMakeFiles.txt around line 40 could use some cleanup with 
regard to the placement of the endif. If the STATIC option is used it 
quietly ignores the failed to detect condition.

I added:

message(STATUS "GG 11: wxWidgets_FOUND = ${wxWidgets_FOUND}")
message(STATUS "GG 20: wxWidgets_LIBRARY_DIRS = ${wxWidgets_LIBRARY_DIRS}")
message(STATUS "GG 21: wxWidgets_LIBRARIES = ${wxWidgets_LIBRARIES}")

to the cmake find function around line 930,

Do you have any wxWidget environment variables set?

I erased all prior wxWidget stuff and installed from source into the 
system include and lib64 directories.

After fudging the wxWidget_FOUND in the prusa code it still fails in the 
Make. Need to go for a while. will post the error in a few hours.

stefanct, 2020-10-19:
The error messages of cmake are really unhelpful. I had the same problem and for me the `-DSLIC3R_GTK=3` was the key (building on Debian 10 Buster). I was looking at https://github.com/prusa3d/PrusaSlicer/blob/master/doc/Dependencies.md and that's not mentioned there (it is in the main linux docs). Maybe add a hint there too since it mentions the `-DSLIC3R_WX_STABLE` flag as well?