prusa3d/PrusaSlicer #2869: [Feature Request] Add Print Speed to Filament Overrides

sgano, 2019-09-05:
### Version
PrusaSlicer 2.1.0-rc

### Operating system type + version
All.  
MacOS v10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa MK2

### Behavior
Is this a new feature request?  YES

The "Filament Overrides" are a great addition to v2.1 - thanks!

I would find it very useful to also have overrides for the print speed as well.

This would simplify the problem I have where certain filaments need to print at different speeds on the same printer.   For instance polyalchemy elixir PLA filament needs to be printed at slower speeds to produce a shiny finish.    Also some brands of PETG print better at slower speeds.   Overriding these printer settings based on a filament type will resolve having to duplicate and modifying the printer settings config for each different filament type that needs these changes.

Print speed was mentioned as a candidate in the v2.1.0 release notes.  I didn't see a formal feature request to add it since that point so that is why I submitted this request. 

#### Project File (.3MF) where problem occurs
NA

AmyTheCute, 2019-09-17:
I'd like to add that this would be amazing. I have a different printer profile for TPU that sets everything to 30mm/s and I sometimes forget to change them back or change them at all.

and v2.1 is really amazing :) Thanks!

bubnikv, 2019-09-17:
You can already set the maximum volumetric speed at the filament profile. Do you need more detailed control over the print speed?

AmyTheCute, 2019-09-17:
> You can already set the maximum volumetric speed at the filament profile. Do you need more detailed control over the print speed?

Yes. because there are filaments like TPU that are bound by other things rather than the hotend's heat output. and all that. Although If i"m being honest this is pretty much purely a TPU thing

alexwhittemore, 2020-01-21:
> Although If i"m being honest this is pretty much purely a TPU thing

I'm not so sure. I came here because I'm having trouble printing ABS - I HAVE to slow down to 70% of my normal "detail" profile print speed to handle the detail on the latest i3  angled part cooling fan shroud. I'm not sure of the best way to handle that, but "ABS-specific printer profile" seems kludgy.

Th3M1k3y, 2020-04-11:
I would love to have this too.

I love how print settings, filament settings and printer settings are split up. However what I keeps finding myself having to do, is to have multiple "Print Settings" for the same settings, but with different speeds because of different filaments.

An example is that currently I am printing a test in PLA to get it finished fast. I got a fast pla setting under "Printer Settings", a PLA setting under "Filament Settings", and then the printer in "Printer Settings" defining the printer I want to use to print it on. When I got the model to fit my needs, I am going to print it in PETG so better be able to handle the heat it is going to be subjected to. But when I am going from PLA to PETG I have to make a new "Print Settings" setting, just to get the speed down to the speed I can print PETG at, and I need a PETG setting under "Filament Settings too".
If there was an option to do "Speed Overrides" just as there is "Filament Overrides" in the "Filament Settings" tab, I would only have to change the filament from PLA to PETG, and then I could hit slice and print, without having multiple profiles for the same print setting, but with different filaments.

haplm, 2020-09-01:
> I love how print settings, filament settings and printer settings are split up. However what I keeps finding myself having to do, is to have multiple "Print Settings" for the same settings, but with different speeds because of different filaments.

The same here, I would love to be able to set speed overrides based on the filament

driph, 2020-10-29:
The filament overrides are very helpful, and adding print speed overrides to the section would be a welcome addition.

neophyl, 2020-10-29:
If all you want to do is scale the speed back by a set % then cant you add a M220 S50 into the filament custom gcode section to scale back to 50% speed given the example above ?  That way you can set filaments to use a % of your normal speeds.  

I know the M220 is the Marlin gcode but I would think the other flavours would also contain something similar ?

driph, 2020-10-29:
> If all you want to do is scale the speed back by a set % then cant you add a M220 S50 into the filament custom gcode section to scale back to 50% speed given the example above ? That way you can set [...]

I believe that also modifies the extruder speed, which would effect retraction, etc.

CocoaPriest, 2020-12-18:
Any updates on this? I would love to have this too.

martinlong1978, 2021-01-29:
I can do this easily in Cura with a plugin. It's absolutely critical for printing TPU, and I'm surprised it isn't in PrusaSlicer.

lukdean, 2021-12-02:
Has this feature been implemented, if so it would be amazing. If not, how are other people working around it? Anyone have any details if this will be an implemented feature and when?

bubnikv, 2021-12-02:
You can limit print speed with maximum filament volumetric flow.

haplm, 2021-12-02:
True, but that's not entirely practical, as it will cap all different speeds at one maximum. It would be much better to be able to define '% of speed decrease' against the print profile, which would scale all the speeds accordingly (not cap them at one limit, which might be too high for say perimeters)

barkh, 2022-04-04:
Volumetric slow did not solve it, in PETG you want limit actual speed - if you go too quick, it will start to acumulate at nozzle. And this speed is almost independend on flow - same speed for 0.1 layers as for 0.4 layers - this is 4 times higher flow, but same max speed...

NyxCode, 2023-03-18:
I like @haplm suggestion of adding a "% speed" override. Additionally, a "max speed" override would be usefull. Then, the final speed for a feature would be `min(max_speed_override, percent_speed_override * speed)`

TomFrost, 2023-04-04:
% speed override isn't enough. For example, for PETG to stick to certain print surfaces, a first layer speed of 10-20 may be necessary. After that layer, infill may print fine at similar speeds as PLA, but perimeters may need to print slower so they can cool to the appropriate glossy or matte finish desired. And as discussed, a maximum cap is even further from this goal.

There's no real benefit to trying to simplify this down to fewer options when Prusaslicer already does a great job at bucketing advanced and expert level settings to keep the UI approachable for each audience. The ability to override every individual speed setting at the expert level per filament is absolutely necessary for fine-tuning the best results.

trebory6, 2023-04-20:
Adding my voice to this as well. I have a troublesome Silk PLA that needs to go slower than my other PLA and even other silk PLA to prevent stringing and preserve details.

I would like to see this as well.

markaabo, 2023-12-04:
This is still a thing we are wanting and waiting for. its been since 2019 and still no update for this simple feature

Th3M1k3y, 2023-12-04:
> This is still a thing we are wanting and waiting for. its been since 2019 and still no update for this simple feature

Just this weekend I wish I had the overrides. Printing the same parts, but with different filaments. It would make so much sense to only change the filament profile, as that was the only thing that was different between the prints.

Gallion, 2023-12-31:
TPU is better printed at a constant speed for everything, meaning that adding a simple speed percentage override won't do. There will still be a need to create additional print profiles only for TPU.

Th3M1k3y, 2023-12-31:
> TPU is better printed at a constant speed for everything, meaning that adding a simple speed percentage override won't do. There will still be a need to create additional print profiles only for TPU.

Would be needed to be able to set both mm/s and as a percentage, so it would be as flexible as possible.

markaabo, 2024-01-02:
> Just this weekend I wish I had the overrides. Printing the same parts, but with different filaments. It would make so much sense to only change the filament profile, as that was the only thing that [...]

This is exactly the reason this needs to be added, really streamlines the printing setup and makes it so much easier for users! @bubnikv

ScriptUp, 2024-02-14:
Hi. I also want speed override future for filament.

bovoro, 2024-07-17:
mid 2024 and I'm still waiting for this feature, and even more so now that the mmu3 is becoming popular, I had to print in 4 colors, 2 Silk and 2 normal PLA, and the only botched solution, reduce the printing speed on the LCD screen, to 80%.

markaabo, 2024-07-17:
@bubnikv @enricoturri1966 @YuSanka @tamasmeszaros
Sorry for the tag, but we need a response

Could we please get this implemented soon, the ticket was opened in 2019!

And as you all can see it's a feature competitors has, and also wanted by alot of people, and it would make things way easier

So please respond, and either give it a closure as not gonna do or respond with some time line or plan or just something we have been waiting ALONG time

martinlong1978, 2024-07-17:
Guys, just because it's been open for a long time doesn't mean it's going
to be done.

There is a volumetric setting per filament which actually does the job
pretty well... You just need to understand how to use it. I was originally
a big advocate for this change, but since I've found the volumetric
setting, it really does everything that's needed.

I suspect that the Devs just don't see this change as necessary because a
suitable feature already exists.

markaabo, 2024-07-17:
> Guys, just because it's been open for a long time doesn't mean it's going to be done. There is a volumetric setting per filament which actually does the job pretty well... You just need to [...]

I get your point but wouldn't it be better for the devs to then close this ticket and mark it as not being implemented so the people know it's been looked at and people can refer to the closed ticket for future simuler suggestions.

Like one reply with "sorry this is not gonna be implemented" and then closing the ticket would take under 30 seconds and be better for all than leaving it open without reply

markaabo, 2025-03-02:
Well, love when a issue is unresolved for years, and the devs just ignore it, even thought it would be so simple and not time wasting to comment stuff like if its not going to be added at this time or something.

See you guys in 2050

dunkmann00, 2025-12-18:
I'll add another use case for this that I don't think was specifically mentioned and that `Max Volumetric Speed` does not really help with. 

I wanted to adjust just the first layer speed for a filament I have and it would be great to set it on a filament profile. Setting it on a print profile is error prone, as I will inevitably forget to change that setting on a project at some point. Changing `Max Volumetric Speed` will slow down everything and I don't want to do that.

So here's hoping for some flavor of this PR being implemented 🤞

closed 2026-09-29 by prusaslicer-bot[bot]