prusa3d/PrusaSlicer #2872: PS slicing toooooo slow

r0xANDt0l, 2019-09-06:
### Version
_Version of PrusaSlicer used goes here_ 2.0.0-rc2

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3, Marlin 2.0

### Behavior
* _Describe the problem_ Slicing very slow
* _Steps needed to reproduce the problem_ slice any big object (or small)
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_1 minute waiting for GCODE
* _Actual Results_ +5 minute waiting for GCODE
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

cggorman, 2019-09-06:
I see this complaint often enough that there must be SOMETHING to it, but my mid-range Dell Inspiron laptop slices 20+ MB plates in less than a minute.  Takes longer to walk the SD card across the room and select the file to print.

lukasmatena, 2019-09-06:
@r0xANDt0l 
Could you please enclose the 3MF? Quoting from the issue report template that you yourself posted as you did not even delete it:
> Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)

There is not much we can do without it. I can only guess that you have enabled something in Print Settings->Layers and Perimeters->Quality which slows down slicing. You can try to retest in PrusaSlicer2.1.0-rc, which should have improved performance of avoid crossing perimeters, if that is the problem.

r0xANDt0l, 2019-09-06:
[MakerSignPS.zip](https://github.com/prusa3d/PrusaSlicer/files/3584313/MakerSignPS.zip)

r0xANDt0l, 2019-09-06:
[MakerSignPS.zip](https://github.com/prusa3d/PrusaSlicer/files/3584316/MakerSignPS.zip)

Jebtrix, 2019-09-06:
Its the usual Avoid Crossing Perimeter quality option causing that.

cggorman, 2019-09-06:
My laptop slices this 3mf in 59 seconds.

JohnOCFII, 2019-09-06:
With a MK3S printer printer profile, Prusament PLA as a selected filament, and a 0.15mm layer height and 15% gyroid infill, my Mac sliced this in under 5 seconds without "Avoid Crossing Perimeters" selected, and 35 seconds with it selected.

aeschrud, 2020-02-10:
I'm having the same issue with the Prusa Slicer V 2.1 and an AMD Ryzen 3700x. It does almost all slicing in seconds and takes minutes to finish the process (Calculating time and costs I guess?)

lukasmatena, 2020-02-11:
@aeschrud Please, read the preceding discussion. There is not much we can do about it if you don't provide a 3MF.

aeschrud, 2020-02-12:
> @aeschrud Please, read the preceding discussion. There is not much we can do about it if you don't provide a 3MF.

@lukasmatena sorry about that, I didn't see it before posting. My bad.

funny enough I wanted to upload a file I just printed yesterday, but today it seemed to take just a few seconds to slice. I'm puzzled.
If it happens again I will post the afm-file that takes long to slice, so you can have a look at it.
Thank you in any case.

Thelvaen, 2020-06-24:
> I'm having the same issue with the Prusa Slicer V 2.1 and an AMD Ryzen 3700x. It does almost all slicing in seconds and takes minutes to finish the process (Calculating time and costs I guess?)

Yeah I think the time calculation could be optimized, maybe by allowing it to use multiple cores?

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

[image]

closed 2020-12-11 by bubnikv

fpvdoc, 2021-01-24:
I got the same problem. The attached 3mf takes at least 5 minutes too slice. Thats super annoying since changing colors needs to be done often and quick.
[Design2.zip](https://github.com/prusa3d/PrusaSlicer/files/5861373/Design2.zip)