prusa3d/PrusaSlicer #2873: "Place on Face" floats object and reports "can't print" error.

GaryBoone, 2019-09-06:
### Version
PrusaSlicer-2.1.0-rc+-201909041313

### Operating system type + version
MacOS 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
* _Describe the problem_

Using "Place on Face" appeared to properly position the object, but an error popped up that read "Empty layers detected, the output would not be printable." Checking the sliced view, it is clear that no layers would be printed until layer 5. It's floating.

* _Steps needed to reproduce the problem_

1. Load the part by either:
    • Open the attached project and re-orient the part to some random position.
    or 
    • Download https://www.thingiverse.com/thing:911891 and load the file 'Saturn V - Stage 1 Tanks- SCALED-2.STL'. It's in the compressed folder 'Saturn V - STAGE 1'.

2. Select the "Place on Face" tool and then click on the bottom/open face. The object is bowl-shaped, so this step attempts to place the bowl upside-down so the lip of the bowl is on the build plate. 

* _Expected Results_

The object should reorient correctly and the slicing should complete successfully.

* _Actual Results_

The object appears to reorient correctly, but an error message appears that says, 

```
Empty layers detected, the output would not be printable.
Object name: Saturn V - Stage 1 Tanks- SCALED-2_repaired.obj
Print z: 1.000000

This is usually caused by negligibly small extrusions or by a faulty model. Try to repair the model or change its orientation on the bed.
```

_Notes_
• This file appeared to work correctly in version 2.0.1. At least, it did not produce the error message.

• Using File | Repair STL file... created a new file with the same problem.

• What's unusual about this file is that the lip of the bowl tapers to a sharp edge. So there's very little contact with the build plate. 

• I was unable to lower the object manually by adjusting its location Z. It reset to the position floating above the build plate. This problem may be another bug.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[plate_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3582121/plate_test.3mf.zip)

Jebtrix, 2019-09-06:
With thin layer detection is gets closer. That base edge is razor thin. Its not really surprising the slicer thinks volume is too small or has no volume at 0.0mm but I guess the slicer just has to do its best there regardless. Practically speaking that is not a model well suited for 3D printing with that base profile imho.

bubnikv, 2019-09-06:
The problem with the thin bottoms is, no perimeter fits in there, therefore an empty layer emerges resulting in the error message you experienced.

Likely the best solution is to print your object on raft, if you want to print it in that orientation, or to use the cutting tool to trim a bit of the object to make the base of the object printable.

FidelCapo, 2019-09-06:
Bottom of this object is too thin. If the object is cut at a height of 0.2mm, the slicing will be OK.
[image]

GaryBoone, 2019-09-07:
@Jebtrix, @FidelCapo: Yes, it seems to be the thin bottom. The bug being reported here is that the Slicer can't correctly manage it. 

@bubnikv: Thanks, good suggestion. I tried printing with a raft, but the object then floated above the raft. Same bug!

@FidelCapo: Cutting at 0.2 sounds like a good workaround. I also found that I could use Cura to simply lower the Z location by 0.5mm and then it worked (in Cura).  However, I couldn't lower it in PrusaSlicer. The setting would change back. As I mentioned above, that also seems to be a bug. 

What I think is the main bug here is that PrusaSlicer doesn't just automatically do the lowering. The feature is "Place on Flat" but it doesn't do that in this case. Instead it floats it. As a user, my expectation is PrusaSlicer will figure out where the material will go when sliced, then place the result on the build plate or raft.

If the devs believe that the behavior has to be this way for some reason, then I would suggest that at least the error box should be clearer in explaining what the problem is.  And it suggest using the Cut feature to ensure a sufficient cross section of the object intersects with the build plate.

bubnikv, 2021-08-31:
> Cutting at 0.2 sounds like a good workaround. I also found that I could use Cura to simply lower the Z location by 0.5mm and then it worked (in Cura). However, I couldn't lower it in PrusaSlicer. The value would change back. As I mentioned above, that also seems to be a bug.

This will be allowed in PrusaSlicer 2.4.0-alpha1.

bubnikv, 2021-09-23:
> Cutting at 0.2 sounds like a good workaround. I also found that I could use Cura to simply lower the Z location by 0.5mm and then it worked (in Cura). 

Enabled in PrusaSlicer 2.4.0-alpha1.
Closintg.

closed 2021-09-23 by bubnikv