prusa3d/PrusaSlicer #2875: Problem with the context menu in the object list

Lato80, 2019-09-06:
### Version
2.1.0-rc+

### Operating system type + version
macOS 10.14.6

### 3D printer brand / version + firmware version (if known)
Custom

### Behavior
After opening an STL file, the context menu for the object cannot be opened in the object list.
If you try to open the context menu in an empty area of the object list, the application crashes.

FidelCapo, 2019-09-06:
Thank you for report. You are right I can reproduce crash but I can open context menu when I select the object by left click and then do right click on the object.

Lato80, 2019-09-06:
I have tested this and if I select the object before, then the context menu could be opened once, but afterwards this workflow didn't work anymore and if I now try to open the context menu in the empty area, then also the program doesn't support anymore.
I closed and reopened PrusaSlicer, but now the workflow doesn't work anymore.

vojtechkral, 2019-09-06:
This should be fixed with the above commit, closing. Let us know in case the problem is still there. Thanks!

closed 2019-09-06 by vojtechkral