prusa3d/PrusaSlicer #2876: Missing parts of the print after slicing

roblatour, 2019-09-06:
### Version
_Version of PrusaSlicer used goes here_

2.1.0 RC-rc (and earlier)

### Operating system type + version
W10 64bit

### 3D printer brand / version + firmware version (if known)
PRUSA i3 mk3s

### Behavior
Create a model in fusion 360 of a box to be printed
move stl file to slicer
view shows entire print (including a bar near the top of the box used to secure the lid)
sliced version does not show the bar
printed version does not show the bar

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs

The following zip file contains:
the auto fusion 360 model, the stl file, 3mf  file, and 4 screenshots showing the problem

[slicer_printer_problem.zip](https://github.com/prusa3d/PrusaSlicer/files/3583644/slicer_printer_problem.zip)

FidelCapo, 2019-09-06:
This model probably contains some errors. I tried to repair the model using the Netfabb service (right click on model and Fix through the Netfabb) and after this repair I think that the model slicing correctly.

roblatour, 2019-09-07:
Yes, that seems to have fixed it - at least in the slicer.  I'll run a print, but am very hopeful.  Thank you.

closed 2021-03-05 by roblatour