prusa3d/PrusaSlicer #2877: Bed size of 8000x8000 mm close prusa slicer

Avatarsia, 2019-09-06:
### Version
_Version of PrusaSlicer used goes here_ 
2.1.0rc+win64-201909041321

Windows 10

if i enter the bed size in the shape menu of 8000x8000 mm slicer close without showing an error

_Is this a new feature request?_

no

Jebtrix, 2019-09-07:
* Win7 64bit - Crashes
* Arch Linux - Works fine

@Avatarsia are your 3D printing house structures? :smile:

Avatarsia, 2019-09-08:
@Jebtrix no not in general, i just had a big project requesed and want a +- overview about the costs and time. 
the total length were 6 m, so my idea was to increase the printer in virtual and see how much filament i would need if i print it in one piece

FidelCapo, 2019-09-09:
Slicer crash when bed size is bigger then 2142mm. Tested on Mac and Win.
Created JIRA-1010.

bubnikv, 2020-04-01:
We are going to limit the print bed to 1.2m x 1.2m. This limit is due to the resolution of 32bit fixed point arithmetics. One saves around 7% RAM and that may make a difference between slicing and not slicing on low end laptops.

YuSanka, 2020-04-01:
The print bed is limited. Close

closed 2020-04-01 by YuSanka