prusa3d/PrusaSlicer #2878: Prusaslicer warning ends up in a loop with per model settings (infill@100%).

mebrein, 2019-09-06:
### Version
2.1.0-rc+win64-201909041321

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
I3 MK3S

### Behavior
When having multiple models infill @20%, when I add a per/model setting of 100%, PrusaSlicer warns about Gyroid cannot be used at 100%. When pressing cancel the message appears again and it ends in a loop. Have to kill process to end PrusaSlicer.
This is not a new feature request.

It can be reproduced without any .3mf file. Just add multiple models.

bubnikv, 2019-09-16:
This issue will be fixed in PrusaSlicer 2.1.0 final. Closing.

closed 2019-09-16 by bubnikv

haarp, 2019-09-27:
@bubnikv @YuSanka 

I am using 2.1.0 final and I'm still experiencing this bug.

Repro variant 1:

1. Add infill settings to object: Pattern = Gyroid
2: Add infill setting to object: Density = 100%
3. -> Switches to rectilinear
4. Remove pattern infill setting
5. -> Infinite loop triggered

Repro variant 2:

1. Add **only** Fill density setting to object
2. Set it to 100%
3. -> Infinite loop triggered

Jebtrix, 2019-09-27:
@haarp Very strange. 2.1.0 final and latest git works fine for me across windows, linux arch repo and appimage.

haarp, 2019-09-28:
Ok, so turns out the 2.1.0 sources downloaded as of today do not face this problem.

Prusa, @bubnikv did you change an already released archive? Please never **ever** do that! Not only do you cause massive confusion on supposedly fixed bugs (such as here), you're also ruining the Linux packages I maintain which rely on the archive hash to verify downloads.

bubnikv, 2019-09-30:
> Prusa, @bubnikv <https://github.com/bubnikv> did you change an already
released archive?

I have moved the label before public releasing the software on github and
everywhere else.