prusa3d/PrusaSlicer #2879: 2,1.0-rc build error in admesh (unusual architecture only)

jasontibbitts, 2019-09-06:
### Version
2.1.0-rc

### Operating system type + version
Fedora 32, building on an IBM s390x mainframe which is a rare big-endian 64 bit architecture.

I know this is a bit out there and that most maintainers don't particularly care about alternative architectures such as this even though they often expose certain assumptions made on , and I'll probably work around this by simply not building on s390x, but 2.0.0 built fine.  It looks like there were some pretty big changes in that source between the two versions, and it does look like there is endian-specific code in there (`#if BOOST_ENDIAN_LITTLE_BYTE` and such) directly around the error.  And it does look like `i` is simply undeclared in context.  Unfortunately simply declaring it as int doesn't get things to compile (though the build fails differently) and I don't really speak C++ or know the code well enough to know why.

```
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0-rc/src/admesh/stl_io.cpp: In function 'bool stl_write_binary(stl_file*, const char*, const char*)':
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0-rc/src/admesh/stl_io.cpp:154:7: error: 'i' was not declared in this scope
BUILDSTDERR:   154 |  for (i = 0; i < stl->stats.number_of_facets; ++ i) {
BUILDSTDERR:       |       ^
BUILDSTDERR: make[2]: *** [src/admesh/CMakeFiles/admesh.dir/build.make:105: src/admesh/CMakeFiles/admesh.dir/stl_io.cpp.o] Error 1
```

bubnikv, 2019-09-09:
Thanks, it should be fixed with 42e41dae0420e40f8386003a97b025f94992e167
It is a regression, which happened during refactoring of the admesh to a C++ way of safer memory allocation, C++ way of error logging etc.

jasontibbitts, 2019-09-09:
Thanks.  I'm embarrassed I didn't figure that out for myself.

jasontibbitts, 2019-09-13:
Just had a chance to test and unfortunately it still doesn't work.  Let me just dump all of the output from building admesh because frankly I can't tell what's actually a problem and what's just the compiler being picky.

```
cd /builddir/build/BUILD/PrusaSlicer-version_2.1.0/Build/src/admesh && /usr/bin/c++  -DBOOST_LOG_DYN_LINK -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxUSE_UNICODE -I/builddir/build/BUILD/PrusaSlicer-version_2.1.0/src -I/builddir/build/BUILD/PrusaSlicer-version_2.1.0/Build/src/platform -I/builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/clipper -I/builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/polypartition -isystem /usr/include/eigen3  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=zEC12 -mtune=z13 -fasynchronous-unwind-tables -fstack-clash-protection -fext-numeric-literals -Wall -Wno-reorder -DNDEBUG -fPIC   -Werror=return-type -Wno-unknown-pragmas -std=gnu++11 -o CMakeFiles/admesh.dir/stl_io.cpp.o -c /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp
BUILDSTDERR: In file included from /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:30:
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl.h: In member function 'void stl_stats::reset()':
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl.h:93:50: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'struct stl_stats' with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
BUILDSTDERR:    93 |  void reset() { memset(this, 0, sizeof(stl_stats)); this->volume = -1.0; }
BUILDSTDERR:       |                                                  ^
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl.h:91:8: note: 'struct stl_stats' declared here
BUILDSTDERR:    91 | struct stl_stats {
BUILDSTDERR:       |        ^~~~~~~~~
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp: In function 'bool stl_write_binary(stl_file*, const char*, const char*)':
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:155:35: error: no match for 'operator+' (operand types are 'std::vector<stl_facet>' and 'size_t' {aka 'long unsigned int'})
BUILDSTDERR:   155 |   memcpy(buffer, stl->facet_start + i, 50);
BUILDSTDERR:       |                  ~~~~~~~~~~~~~~~~ ^ ~
BUILDSTDERR:       |                       |             |
BUILDSTDERR:       |                       |             size_t {aka long unsigned int}
BUILDSTDERR:       |                       std::vector<stl_facet>
BUILDSTDERR: In file included from /usr/include/c++/9/bits/stl_algobase.h:67,
BUILDSTDERR:                  from /usr/include/c++/9/bits/char_traits.h:39,
BUILDSTDERR:                  from /usr/include/c++/9/ios:40,
BUILDSTDERR:                  from /usr/include/c++/9/ostream:38,
BUILDSTDERR:                  from /usr/include/boost/log/trivial.hpp:19,
BUILDSTDERR:                  from /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:26:
BUILDSTDERR: /usr/include/c++/9/bits/stl_iterator.h:423:5: note: candidate: 'template<class _Iterator> std::reverse_iterator<_Iterator> std::operator+(typename std::reverse_iterator<_Iterator>::difference_type, const std::reverse_iterator<_Iterator>&)'
BUILDSTDERR:   423 |     operator+(typename reverse_iterator<_Iterator>::difference_type __n,
BUILDSTDERR:       |     ^~~~~~~~
BUILDSTDERR: /usr/include/c++/9/bits/stl_iterator.h:423:5: note:   template argument deduction/substitution failed:
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:155:37: note:   mismatched types 'const std::reverse_iterator<_Iterator>' and 'size_t' {aka 'long unsigned int'}
BUILDSTDERR:   155 |   memcpy(buffer, stl->facet_start + i, 50);
BUILDSTDERR:       |                                     ^
BUILDSTDERR: In file included from /usr/include/c++/9/bits/stl_algobase.h:67,
BUILDSTDERR:                  from /usr/include/c++/9/bits/char_traits.h:39,
BUILDSTDERR:                  from /usr/include/c++/9/ios:40,
BUILDSTDERR:                  from /usr/include/c++/9/ostream:38,
BUILDSTDERR:                  from /usr/include/boost/log/trivial.hpp:19,
BUILDSTDERR:                  from /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:26:
BUILDSTDERR: /usr/include/c++/9/bits/stl_iterator.h:1219:5: note: candidate: 'template<class _Iterator> std::move_iterator<_IteratorL> std::operator+(typename std::move_iterator<_IteratorL>::difference_type, const std::move_iterator<_IteratorL>&)'
BUILDSTDERR:  1219 |     operator+(typename move_iterator<_Iterator>::difference_type __n,
BUILDSTDERR:       |     ^~~~~~~~
BUILDSTDERR: /usr/include/c++/9/bits/stl_iterator.h:1219:5: note:   template argument deduction/substitution failed:
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:155:37: note:   mismatched types 'const std::move_iterator<_IteratorL>' and 'size_t' {aka 'long unsigned int'}
BUILDSTDERR:   155 |   memcpy(buffer, stl->facet_start + i, 50);
BUILDSTDERR:       |                                     ^
BUILDSTDERR: In file included from /usr/include/c++/9/string:55,
[... cut ...]
BUILDSTDERR: /usr/include/c++/9/bits/stl_iterator.h:996:5: note:   template argument deduction/substitution failed:
BUILDSTDERR: /builddir/build/BUILD/PrusaSlicer-version_2.1.0/src/admesh/stl_io.cpp:155:37: note:   mismatched types 'const __gnu_cxx::__normal_iterator<_Iterator, _Container>' and 'size_t' {aka 'long unsigned int'}
BUILDSTDERR:   155 |   memcpy(buffer, stl->facet_start + i, 50);
BUILDSTDERR:       |                                     ^
BUILDSTDERR: make[2]: *** [src/admesh/CMakeFiles/admesh.dir/build.make:105: src/admesh/CMakeFiles/admesh.dir/stl_io.cpp.o] Error 1
```

jasontibbitts, 2019-09-18:
Any chance someone with the necessary privileges could repoen this?  Or should I just open a new issue?  Unfortunately it's left me with a dilemma as I'd like to push an updated package to all Fedora releases, but that would necessitate leaving out an architecture which would then be stuck on an old version.

bubnikv, 2019-09-19:
Sorry for the trouble.

Please patch the big endian build with 

@@ -151,8 +151,8 @@ bool stl_write_binary(stl_file *stl, const char *file, const char *label)
 	memcpy(buffer, &stl->stats.number_of_facets, 4);
 	stl_internal_reverse_quads(buffer, 4);
 	fwrite(buffer, 4, 1, fp);
-	for (size_t i = 0; i < stl->stats.number_of_facets; ++ i) {
-		memcpy(buffer, stl->facet_start + i, 50);
+	for (const stl_facet &facet : stl->facet_start) {
+		memcpy(buffer, &facet, 50);

I will commit it into the dev branch as of now. The dev branch will soon be merged over master before the master goes merged to stable.

bubnikv, 2019-09-19:
```
@@ -151,8 +151,8 @@ bool stl_write_binary(stl_file *stl, const char *file, const char *label)
 	memcpy(buffer, &stl->stats.number_of_facets, 4);
 	stl_internal_reverse_quads(buffer, 4);
 	fwrite(buffer, 4, 1, fp);
-	for (size_t i = 0; i < stl->stats.number_of_facets; ++ i) {
-		memcpy(buffer, stl->facet_start + i, 50);
+	for (const stl_facet &facet : stl->facet_start) {
+		memcpy(buffer, &facet, 50);
```

closed 2019-09-20 by bubnikv

jasontibbitts, 2019-09-23:
Even with this there's still a build error, but fortunately I can fix this one myself.  I submitted #2981.

commit 94212fa2 (2019-09-19): Another fix of admesh on big endian architectures, fixes #2879