prusa3d/PrusaSlicer #2880: 2.1-rc no export model showen

Avatarsia, 2019-09-06:
### Version
_Version of PrusaSlicer used goes here_
2.1-rc
 - win64

after slicing the object got shown for  second and then disappears.
file in the attachment

[NutCracker_Plunger.zip.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/3585247/NutCracker_Plunger.zip.amf.zip)

Jebtrix, 2019-09-07:
Any model I drop in that project disappears. I adjusted every single value there is to match a Prusa default profile. Same result. I'm stumped.

Jebtrix, 2019-09-07:
Devs, editing the .amf manually I got it to work. It's something to do with the printer profile and the bed model. When I change it to ie Prusa i3 MK2 the bed model shows and the slice preview acts normal.

Jebtrix, 2019-09-08:
This project may work on v2.0 for Windows but Linux v2.0 build it fails the same way. If I just change the project file's empty Printer > Before layer change G-code to `G92 E0.0` I can something to render.

[image]

Avatarsia, 2019-09-08:
it shouldnt have a 2nd color, its a 1 color print, i'm a bit confused

Jebtrix, 2019-09-08:
You referring to my reference to the two color problem issue? It just shared a preview problem and a similar make the profile more like a Prusa printer fix. So a very loose relation :)

Yours was actually an easy fix, I have no idea how I missed it. Too many things at once today I think. Your extrusion multiplier was 0, should be close to 1.

Avatarsia, 2019-09-08:
oh lol, yeah that was simple, normally i use 0.9, shouldnt be 0 
thanks for your eyes :)

closed 2019-09-08 by Avatarsia