prusa3d/PrusaSlicer #2882: Plater panel combobox icons display issue

Jebtrix, 2019-09-07:
### Version
version_2.1.0-rc-5-gfec111302

### Operating system type + version
Arch Linux
* -DSLIC3R_STATIC=0 
* -DSLIC3R_FHS=1 
* -DSLIC3R_WX_STABLE=1 
* -DSLIC3R_GTK=3 
* -DSLIC3R_PCH=0 

### Behavior
Icon images are not showing transparent background outside their native size.

[image]
width are getting a black background instead of transparent.

Jebtrix, 2019-09-07:
Using latest wxWidgets 3.1.2 does **_not_** have this problem.

VanessaE, 2020-12-15:
I can confirm this as well, at Prusaslicer commit 26755eb1d1fa4ed35ef0facc0aa80a1f2996a756.

I have Wx 3.1.3 on my system (Debian buster+sid).

[image]

The garbage behind the icons appears to be random imagery from elsewhere on my desktop (probably an uninitialized image buffer or something?) filling-in where the icon looks like it was meant to be transparent.

Build log:
<details><summary>Show log...</summary>

```make
vanessa@rainbird:~/RepRap/PrusaSlicer$ ~/Scripts/build-Prusaslicer.sh 
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: OFF
-- CMAKE_PREFIX_PATH: (default)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (DBUS) does
  not match the name of the calling package (DBus).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindDBus.cmake:59 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:169 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DBUS: /usr/include/dbus-1.0;/usr/lib/x86_64-linux-gnu/dbus-1.0/include  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.64.0") found components: system filesystem thread log locale regex chrono atomic date_time 
-- Boost::boost exists
-- Found TBB: /usr/include (found version "2020.3")  
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /usr/lib/x86_64-linux-gnu/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
[... 1523 lines cut ...]
/usr/bin/ld: warning: libboost_iostreams.so.1.74.0, needed by /usr/lib/x86_64-linux-gnu/libopenvdb.so, may conflict with libboost_iostreams.so.1.71.0
[100%] Linking CXX executable slic3rutils_tests
[100%] Built target slic3rutils_tests
Symlinking the G-code viewer to PrusaSlicer
Symlinking the resources directory into the build tree
[100%] Built target PrusaSlicer
vanessa@rainbird:~/RepRap/PrusaSlicer$
```
</details>

Contents of the build script:

<details><summary>Show that script...</summary>

```bash
#!/bin/bash

cd /home/vanessa/RepRap/PrusaSlicer/
rm -rf build
mkdir build
cd build
cmake .. 
make -j8
```
</details>

bubnikv, 2020-12-16:
I am sorry, but we cannot reproduce the issue.

In general, we cannot support every combination of libraries. We will only support our binary builds and builds compiled against the static dependencies pulled by our dep build scripts. Do you have such an issue with our binaries or self compiled against the static dependencies pulled by our dep build script?

bubnikv, 2020-12-16:
Naturally if we get a pull request fixing the issue and retested on all other combinations of all possible libraries, we will be more than happy to merge it.

VanessaE, 2020-12-16:
Well if it helps any, I went so far as to wipe my drive and re-install my system completely from scratch, figuring this was one of those conflicts that's dug itself in under too many layers of system cruft.

This is Debian Bullseye/testing (with Sid/unstable available but unused). Right after booting into the newly-installed system for the first time, before installing any software, I did `apt-get autoremove libboost.*`, then did some apt pinning, so that only WxWidgets 3.1.4 and Boost 1.74 can be installed.  Then I began installing my usual stuff.  No conflicts of any kind occurred during the setup.

WxWidgets came from its official codelite.org Debian repository.  Boost is from the standard Debian repository.

That's as clean a slate to start from if there ever was one, plus if I didn't do those steps, I'd get a combination of old and new Boost and Wx packages, which would cause other minor issues (not related to Prusaslicer).

I focus on those two projects because PS uses both, and they're literally the ONLY things that have any question about them, plus of course Wx seems to be central to this graphical glitch.

As for why I compile PS versus using official builds, well, I've pretty much always done that.  After all, if you can, I should be able to also. :smile:  I was not, however, aware of the static dep scripts you mention -- my method is an offshoot of Slic3r, and continued from when Prusaslicer was still called "SlicerPE".

That aside, I went ahead and tried one (PrusaSlicer-2.3.0-beta3+linux-x64-202012121815.AppImage), and its display looks normal, or at least as normal as my previous compiled efforts.

Thing is, this issue is really pretty localized.  ONLY the icons used in the three presets' dropdown menus are broken (both on the Plater screen and in the three presets' tabs).  All other imagery and icons look fine throughout the program, including in regular menus, and on buttons which contain an icon. And besides all that, it's only the transparent region *behind* the icon that's broken -- the actual icon image itself is still there and it's intact -- see here for example:

[image]

Those colored squares match the colors I assigned to the filaments, and are exactly what I should be seeing, garbage behind them notwithstanding.

Anyway, no one's asking for "all combinations" of libraries to work.  I mean, dude, this is Debian.  If there's ANY standards in Linux to be found, they're here.  If a build doesn't come out properly here after all the precautions I took when reinstalling my OS, but it builds for you, then I would suggest that it's your system that's non-standard, and that someone is misusing a Wx feature in a way that works on your system but not mine.  So please, at the risk of sounding bitter, dispense with the hyperbole, it doesn't help anything and just pisses people off.

VanessaE, 2020-12-16:
Since I'm borderline OCD and a little paranoid, :smile: I decided to just take this to its logical conclusion, just to eliminate all possible sources of problems:

I wiped and reinstalled again, this time from a Debian testing minimal network-install image.  That way, I started with nothing but the bare essentials i.e. just the Linux VT/console, network access, apt, basic system tools, and little else.

Ran through my install-setup process (I wrote a not-quite-a-script ages ago to guide me through it, so that there are no mistakes), and this time, literally *nothing* had to be uninstalled/removed to get my system to the exact state I wanted it in.  Naturally I had to also install a graphical environment, but that wasn't anything major, a handful of meta packages like `xfce4`, `lightdm`, etc.  In fact, the whole schmear seems to be a little snappier now, which is a nice bonus.

Point being, there's literally nothing else I can do to establish a known, stable configuration that I could use indefinitely.

Recompiled Prusaslicer again, still the same problem.

lukasmatena, 2020-12-17:
@VanessaE I believe the advise from @bubnikv should be sufficient. Clone PrusaSlicer, build the dependencies using the cmake script in deps directory (which will clone the exact versions that we know will work). Then build PrusaSlicer by its top cmake script, using CMAKE_PREFIX_PATH variable to point it to the built dependencies and set SLIC3R_STATIC=1. The process is described in the [wiki page](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) and it should work. This is how our builds are made, we do not link against any "standard" system libraries provided by our favourite Linux distribution. If you insist on dynamic linkage against system libraries, feel free to do so, but we will not fix issues caused by possible mismatches. We did for a long time, it is a never ending story and we would not do anything else.

> Anyway, no one's asking for "all combinations" of libraries to work.

No, but everyone wants "their" combination to work, which is effectively the same.

> I mean, dude, this is Debian.

And why does it have 3.1.3 wxWidgets installed? Or is it 3.1.4 (which you also mention)? I'd expect the years old 3.0.4 ([link](https://packages.debian.org/buster/libdevel/)) that most distros provide, no?

> someone is misusing a Wx feature in a way that works on your system but not mine

We link against different wx version than you, that is not necessarily a misuse. It would be somewhat easier if wxWidgets authors proclaimed 3.1 series stable and distributions would at least ditch the 3.0, but... they didn't. Not to mention GTK2/3/4 variants.

Do not take it the wrong way, I myself use Linux on my desktop for everyday work. But trying to keep PrusaSlicer working with libraries that someone simply pulls from some repository and builds the software against seems to be futile. And the wxWidgets status does not make it easier. Fix the issue, if you see it - if it works and does not break anything else, we will gladly merge it. But we must also support the other two platforms which are by the way used by vast majority of our users.

VanessaE, 2020-12-17:
> And why does it have 3.1.3 wxWidgets installed? Or is it 3.1.4 (which you also mention)?
> I'd expect the years old 3.0.4 (link) that most distros provide, no?

3.1.x comes from the official WxWidgets repository for Debian from codelite.org, by way of my having added it to `/etc/apt/sources.list`.  In contrast, standard debian.org "Bullseye/testing" repositories provide Wx 3.0.5.

When I posted my initial comment, it was 3.1.3, but in the midst of those 9999 wipe-reinstall cycles :stuck_out_tongue: , it switched 3.1.4 -- probably just due to routine updates in the repository. That's not great timing for sure, but that bit's kinda out of my control.  To be clear, the issue is the same in 3.1.3 and 3.1.4.

It's odd that @Jebtrix has it happening in 3.0.4 and not 3.1.2 (as he mentioned).  This would seem to point the finger not at WxWidgets but at some other library that Prusaslicer's using.

> We link against different wx version than you [...]

Well perhaps you and/or those previously-mentioned scripts do, but for normal building per PrusaSlicer's documentation at https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md, 3.0 is not the default. To wit:

> By default, PrusaSlicer looks for wxWidgets 3.1, this is because the 3.1 version has a number of bugfixes and improvements not found in 3.0. However, it can also be built with wxWidgets 3.0. This is done by passing this option to CMake:
>
> `-DSLIC3R_WX_STABLE=1`
>
> Note that PrusaSlicer is tested with wxWidgets 3.0 somewhat sporadically and so there may be bugs in bleeding edge releases.

Between the above and a different, unrelated graphical issue that happened at one time (ages ago), I switched to 3.1 and simply continued using it.  Until today that is.

> that is not necessarily a misuse

No, I meant whatever the specific feature of Wx that makes icons inside of dropdowns/comboboxes possible.  I mean, transparency is not a new concept. :stuck_out_tongue:   Visually, it just screams "uninitialized memory".

Now all of the above aside, I took one last stab at making it work right, and reinstalled my system yet again (as before, to make sure there are no questions, conflicts, or cruft left behind from before), but this time without blocking-out older Boost, and without enabling the Wx repository.  Thus the system has a mix of Boost 1.71 and 1.74, and Wx is at 3.0.5, as mentioned above.

Building against 3.0.5.1 results in correct icons in the dropdown menus, but then goes slightly wrong in a couple other ways (see #5494 and #5495).

VanessaE, 2020-12-17:
I decided to make one more attempt, this time going back to Debian stable, and not running `apt-get upgrade` until everything I use was installed (instead of doing so early-on).  There must have been some library that broke silently before.

bubnikv, 2021-09-23:
We decided to only support Linux builds compiled against our static dependencies as described here:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

We are open for pull requests fixing issues due to distro dependencies.

closed 2021-09-23 by bubnikv