prusa3d/PrusaSlicer #2883: Supports generating even though not selected

Sanittech, 2019-09-07:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
Mac OS High Sierra (10.13.6)

### 3D printer brand / version + firmware version (if known)
Cr10s Marlin 1.1.9

### Behavior
* Supports generating even though not selected.

Attached are the settings and result of the slice even though it is clearly no supports are selected.
[image]
[image]
[image]

The only way to stop support from generating support by adding a support blocker or enforcer or support settings to stop it.
[image]

#### Project File (.3MF) where problem occurs

[20mm_kockica.zip](https://github.com/prusa3d/PrusaSlicer/files/3586779/20mm_kockica.zip)

Jebtrix, 2019-09-07:
Can't reproduce that on Win7 or Linux. When I first opened that project Supports was set to Everywhere. Setting it to None worked fine.

Sanittech, 2019-09-08:
understood but even when I don't enable support it keeps showing up and enabled.

Jebtrix, 2019-09-08:
Try latest version at (2.1.0rc)[https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.1.0-rc]

rtyr, 2021-02-03:
Likely fixed. I am unable to reproduce in 2.3.0. Closing.

closed 2021-02-03 by rtyr