prusa3d/PrusaSlicer #2884: 2.1.0-rc Wrong color of Extruder 2 in tool print preview

rmoravcik, 2019-09-07:
Hi all,

I'm using PrusaSlicer 2.1.0-rc on Ubuntu Linux 19.10 (alpha)

[image]

I wanted to check an print preview of simple multi material object and slicer is showing paths for both extruders with same color.

[image]

I also noticed this. When slicer is re-generasting an stl, then part of object, that should be printed with second extruder has correct color. And as soon as generation is finished it's changed back to wrong color.

[image]

If I check tool preview in 2.0.0 version of PrusaSlicer, then everything is ok.

[can_cover.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3586805/can_cover.3mf.zip)

Jebtrix, 2019-09-07:
Yeah there is a problem. If i change the printer to a Prusa default ie Mk2S the preview works fine. It may related to issue [#2883](https://github.com/prusa3d/PrusaSlicer/issues/2883). Preview is favoring only Prusa preset printer profiles (and derivatives of) for some reason at the moment.

rmoravcik, 2019-09-07:
I just little bit played with my printer profile and I discovered this. As soon as toolchange_gcode is set, then tool paths in print preview are rendered with the same color. If I clean toolchange_gcode, then tool paths are rendered with right colors. My printer profile has toolchange_gcode set to M600.

Jebtrix, 2019-09-07:
It broke from this commit [e674c58 ](https://github.com/prusa3d/PrusaSlicer/commit/e674c586b0689e2424512228a4927dcfc61ce346).

lukasmatena, 2019-09-08:
It is by design. If you set your toolchange gcode and decide it should do nothing, then it does... well, nothing. This behaviour is explained in the tooltip on the toolchange gcode. There was already a discussion in https://github.com/prusa3d/PrusaSlicer/issues/1530.

Jebtrix, 2019-09-08:
@lukasmatena I read that issue. Can't the slicer imply a T[next_extruder] if it doesn't see it in custom gcode, at least for the preview but stripped out during export?

lukasmatena, 2019-09-08:
We thought about this too and will probably discuss it again tomorrow.

lukasmatena, 2019-09-09:
Commit https://github.com/prusa3d/PrusaSlicer/commit/97a02f4412189215b47bd49f3553da3d7f1cd6cb added a search for a tool-changing command in the custom gcode. In case it is not found, PrusaSlicer will emit it itself just after the custom gcode. I have also updated the tooltip accordingly.

We'll push the fix into 2.1.0 final. I hope it is correct. You can retest the behaviour in upcoming PrusaSlicer2.1.0-rc2 and close the issue if ok.

bubnikv, 2019-09-11:
Implemented in PrusaSlicer 2.1.0-rc2. Closing.

closed 2019-09-11 by bubnikv