prusa3d/PrusaSlicer #2885: "Send G-code" button disappears after making changes to a print/filament settings

nebbywan, 2019-09-07:
### Version
2.1.0-rc commit fec1113020e7a677af83738b17b6867e44b56ec9

### Operating system type + version
Win10 Home 1903

### Behavior
"Send G-code" button disappears after making changes to a print/filament settings and returning to Plater/re-slicing. Only restarting the program seems to fix the issue.

FidelCapo, 2019-09-09:
Can you please describe all the steps needed to reproduce the problem? Video would definitely help in this case. Thank you

nebbywan, 2019-09-09:
Video attached.
[PrusaSlicer-2.1.0-rc+UNKNOWN based on Slic3r 2019-09-09 00-01-31.zip](https://github.com/prusa3d/PrusaSlicer/files/3589050/PrusaSlicer-2.1.0-rc%2BUNKNOWN.based.on.Slic3r.2019-09-09.00-01-31.zip)

nebbywan, 2019-09-09:
So a couple other odd notes. I just tried to make a video of the normal behavior in 2.0.0 and it actually did the same thing. Thought it might be my "Pretty PLA" printer, so I copied the IP/API of my OctoPrint server to the stock MK3S config and it *still* did the same thing. I'll see if this follows to a different computer tomorrow...

FidelCapo, 2019-09-09:
Thank you. Unfortunately I still can't reproduce this issue on win or mac. Can you please send a configuration file or a 3mf file? Thank you

nebbywan, 2019-09-09:
Here's the 3mf and my config folder.
[3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3589116/3mf.zip)
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/3589117/PrusaSlicer.zip)

FidelCapo, 2019-09-09:
Ok, with your profiles I can reproduce this issue.  Thank you.

nebbywan, 2019-09-09:
Awesome! Well...sorta. Let me know if there's anything more I can do to help, and thank you for yours.

FidelCapo, 2019-09-09:
One think what I found. If I open Printer Settings tab, there is no IP and API, but I know you have IP and API in config files. Do you have these settings in Printer settings tab If you try slice object for first time (Send gcode is there) and after making changes in Print settings tab?

nebbywan, 2019-09-09:
Yes, that info is saved in the Printer settings. After I slice the first 
time the send button is there, but as soon as I go back to tweak a print or 
filament setting and reslice it's gone. I've even had it (but with less 
consistency) when just adding modifiers like support blockers on the plater.

FidelCapo, 2019-09-09:
Is IP and API settings gone too if Send gcode button is gone?

nebbywan, 2019-09-09:
It's still there (and tests ok) when the button is gone

FidelCapo, 2019-09-09:
Thank you for info. I hope we find out what's happening here soon.

nebbywan, 2020-10-29:
This is no longer an issue after the GUI updates.

closed 2020-10-29 by nebbywan