prusa3d/PrusaSlicer #2886: MMU - Fan speed is not set properly if there is some content (even ;) in Tool change G-code

rongith, 2019-09-07:
### Version
PrusaSlicer 2.1.0-rc

### Operating system type + version
Windows 10 x64 1903

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
This is in relation with https://github.com/prusa3d/PrusaSlicer/issues/2884
I opened a new issue because it is not only the rendered colors that are impacted.
Put at least one character in Tool change G-code, even a single comment character.

Not only the rendered color of the tool is not set properly, fan speed of the second tool is not set properly, M106 Sx are the with the values of the first tool. Didn't make tests with other tool changes.

lukasmatena, 2019-09-08:
Please see https://github.com/prusa3d/PrusaSlicer/issues/2884#issuecomment-529174859.
Actually, I will close this as a duplicate of #2884.

closed 2019-09-08 by lukasmatena

Jebtrix, 2019-09-08:
I'm confused. Issue #2884 is easy to understand the result but exactly how is filament fan speeds connected to having T[next_extruder] in custom gode?

rongith, 2019-09-08:
@lukasmatena. Then why the volumetric speed of the second tool is correct and not the fan speed?

rongith, 2019-09-08:
Then we can't do multimaterial without MMU  properly with M600 command.

lukasmatena, 2019-09-08:
> Then we can't do multimaterial without MMU properly with M600 command.

You can. Just read the tooltip. I'll comment on the rest when I have time to look into a bit more.

lukasmatena, 2019-09-08:
> but exactly how is filament fan speeds connected to having T[next_extruder] in custom gode?

Fan settings are a filament thing. If you enter the ";" into custom toolchange gcode, there is no toolchange. And the fan commands are the same as if you only printed with the first filament the whole time.

> It's not just the starting fan speed that goes missing, but also any changes due to layer time.

The starting fan speed has not gone missing, as far as I can see. As for the changes due to layer time... Do they really go missing? I just sliced the same model with an MK3 and there are no other fan speed commands, apart from the first one. It seems they are not supposed to be there at all.

> Then why the volumetric speed of the second tool is correct and not the fan speed?

This is a good point. I'd rather say it's the other way round, fan speed is correct and the volumetric speed is not (the preview is supposed to reflect the gcode). But you are right this is not ok. Maybe forcing the Tn command in case it is not found in the toolchange script would solve both the confusion and discrepancies like this.

rongith, 2019-09-08:
> This is a good point. I'd rather say it's the other way round, fan speed is correct and the volumetric speed is not (the preview is supposed to reflect the gcode). But you are right this is not ok. [...]

OK. Don't know what is best to help people pushing the software without knowing what is behind exactly.
I'm happy by adding the Tn command and the know how. You would do 1.documentation or 2.force Tn.

lukasmatena, 2019-09-09:
Commit https://github.com/prusa3d/PrusaSlicer/commit/97a02f4412189215b47bd49f3553da3d7f1cd6cb added a search for a tool-changing command in the custom gcode. In case it is not found, PrusaSlicer will emit it itself just after the custom gcode. This should solve the confusion, problem with the fan speed and problems with the preview.

We'll push the fix into 2.1.0 final. I hope it is correct.

bubnikv, 2019-09-11:
Implemented in PrusaSlicer 2.1.0-rc2.