prusa3d/PrusaSlicer #2887: 2.1.0-rc height range features applied to entire object

ex-nerd, 2019-09-08:
### Version
2.1.0-rc

### Operating system type + version
MacOS 10.14.6

### 3D printer brand / version + firmware version (if known)
not actually sending to printer yet, but it's a custom kossel style delta, Duet controller.

### Behavior
* Using the new height range to change perimeters and infill percentage applies changes to the entire model, in *addition* to the default values used on the rest of the model.
* Reproduction:
  * Add model configured with 2 perimeters, 10% infill, gyroid pattern
  * create a height range from 0-10mm with overrides for perimeters (something large like 6) and infill (some large number like 30%, different pattern than global, e.g. rectilinear)
  * slice and use the layer view to look at all layers
* _Expected Results_
  * from 0-10mm, 6 perimeters, 30% rectilinear infill
  * above 10mm, 2 perimeters and 10% gyroid infill
* _Actual Results_
  * entire model uses 6 perimeters (maybe overlapping with 2) and *both* 10% gyroid infill *and* 30% rectilinear infill
  * [image]

_Is this a new feature request?_  No

#### Project File (.3MF) where problem occurs

Current object I'm dealing with is proprietary, but I will upload a plain one if you are unable to replicate the issue.

Jebtrix, 2019-09-08:
Do you have Print Settings > Advanced > Other > Clip multi-part objects checked? If not try that.

ex-nerd, 2019-09-08:
That seems to have fixed things.  Though I'm actually surprised that the option is even enabled in single-material mode (I've had to switch to single-material print mode before to enable features like material-change-at-layer-height to fake out a "pause" functionality to drop nuts into parts mid-print). Anyway, it still feels like a bug for single-material prints (or even multi-material prints where you might want this feature but *don't* want clipping)

Jebtrix, 2019-09-08:
That feature is on a likely on the UI removal list. You can read devs comments here issue [ #2803](https://github.com/prusa3d/PrusaSlicer/issues/2803)

ex-nerd, 2019-09-08:
Thanks for the info, it was a good read, and explains why the feature was hard to find and not enabled for my setup.

bubnikv, 2019-09-09:
I just checked that we have the mutual clipping enabled even for single
material profiles for the Prusa Research printers.

I know that people bend the software all possible and impossible ways. For
example, https://3dlabprint.com/ bends Simplify3D by overlapping multiple
bodies one over the other, basically producing invalid geometry. If
processed by PrusaSlicer, these parts are merged together. Maybe somebody
somewhere has a use for not enabling the mutual clipping.

lukasmatena, 2024-02-18:
Solved, closing.
The "Clip multi-part objects" was removed in the meantime (in 2.6.0-alpha4).

closed 2024-02-18 by lukasmatena