prusa3d/PrusaSlicer #2888: Exported message ends with ...

francisdb, 2019-09-08:
### Version
`2.0.0+`

After exporting gcode the status bar still shows a `...` (ellipsis) at the end of the message which normally indicates something is still busy. This sentence should end with `.`

[image]

Jebtrix, 2019-09-08:
Every truncation function I've seen in any programming language is ... It's not a sentence its a truncated path. The program does use terms like Generating G-Code... so I do see what your saying. Technically an ellipsis for both is grammatically correct and proper use in their respective context.

Jebtrix, 2019-09-08:
Actually you are correct.

bubnikv, 2021-09-25:
In prusaSlicer 2.4.0-alpha2 the status bar will be gone.

closed 2021-09-25 by bubnikv