prusa3d/PrusaSlicer #2889: Confusing dialog when trying to enable soluble interfaces with support_material_contact_distance > 0.

sampoz, 2019-09-08:
### Version
PrusaSlicer-2.1.0-beta2+linux64-201908250731
### Operating system type + version
Debian 9
### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3s Multi Material 2S
### Behavior
PrusaSlicer gives a bit subversive message when trying to enable soluble supports:
[image]
```
The Wipe Tower currently supports the non-soluble supports only
if they are printed with the current extruder without triggering a tool change
(both support_material_extruder and support_material_interface_extruder need to be set to 0).
\nShall I adjust those settings in order to enable the Wipe Tower?
```

With a bit of spelunking around in the code, it seems that wipe tower is supported, but to use it, one has to set "support_material_contact_distance" to 0 :
[image]

#### Project File (.3MF) where problem occurs
This is not about a project, but print settings present, so no project included

I propose the text to be changed to include the information about the support_material_contact_distance.

bubnikv, 2019-09-09:
What do you propose?

vertigo235, 2019-09-20:
@bubnikv 	I propose allowing the contact z distance to be whatever you want, why is this being enforced at all?  I would like to use breakaway support material as the interface, and that requires z distance slightly greater than 0, this constraint doesn't make sense to me, but maybe I am missing something?

bubnikv, 2019-09-20:
Zero distance was implement in such a way that it always means soluble support interface.

Would you please describe your use case in more detail? What material do you plan as a breakaway support? Why would you print that breakaway support material with zero contact Z distance? Shouldn't you leave at least some gap?

vertigo235, 2019-09-20:
Correct, would like a very small gap, but with this constraint you have to choose no gap or no wipe tower.

vertigo235, 2019-09-20:
There is something that prevents usage of the wipe tower unless you choose 0 distance that I do not understand?  

To clarify, the use case.

I would like to use breakway support material as a support interface (like polysupport or inego 3d450), but I would like to have 0.05 z seperation (not 0), and I would also like to have a purge tower as well as regular PLA as the support structure. 

This doesn't appear possible with this constraint, you would have to choose to have 0 seperation if you want to use purge tower / block. What is the reason for this constraint, why are they mutually exclusive?

firelizzard18, 2020-02-19:
I have two reels of PLA. One is a material I like very much and would like to keep using. One is what I use for test prints and anything else I don't particularly care about.

Slicing my model without supports shows that it uses ~80g of filament. With supports it's 140g. So that's 60g of wasted filament. I do not want to waste 60g of my nice filament. But this constraint is preventing me from changing that.

TheStig23, 2020-03-08:
I completely agree with vertigo235. I have this problem too. My multi-part object has 1 pla body, and 1 soluble body. I want to print the part with pla supports, but the slicer alternates the support layers. Steep pla overhangs do not work on pva soluble supports. Removing this constraint would solve my problem

bubnikv, 2020-03-09:
PrusaSlicer prints the support layers independently from the object layers by default, meaning that the object layers are printed at layer heights independent from the support layer heights. This is an effective strategy if the object / support gap is not a multiple of the object layer height, and this strategy also allows the support columns to be printed with thicker layers thus faster than the object layers.

However, printing the support layers independently from the object layers poses difficulty in handling the filament switches on the wipe tower. We decided to solve the difficulty by two strategies:

1) For normal supports, supports are always printed with the last active extruder, therefore there is no need for filament swap.

2) For soluble support interfaces, the support interfaces are printed at the object layer height, therefore the filament exchange sequence is performed in the same way as the filament exchange sequence for the object materials.

This is an explanation of the current state. One may basically point his finger randomly at any place in slicer's source code and he will certainly find some idea for improvement, however we have to judge the priorities.

matyay, 2020-03-22:
Hello,

I have a potential use case for non-soluble support plus wipe tower. Suppose that you want to print something using materials of different kinds. Like for example PLA and TPU (elastic). These material types require completely different speed / retraction settings etc. Therefore you may want to use only one of them for the support, definitely not both interleaved.

I'm not familiar with implementation details but could the wipe tower be also printed somehow "independently" from objects and support? With the progressing print the slicer has to choose what to print next, it can be either object or support regardless of their layer heights. At that point part of the wipe tower can be printed using adaptive layer height, possibly equal to the Z distance between previous and next tool change position.

spiff72, 2021-01-18:
I just stumbled on this github issue as I was searching for a solution to my issue.

I am experimenting with some new PolySupport breakaway material, and looking at using it with PETG object material (it is better suited to use with PLA, but I am experimenting, as I said earlier).  I have run a few prints where I identified the PolySupport material as soluble, which forces me to use contact distance of zero.  I am finding that the support material still sticks too well to the object, so was going to try a small gap (.05 or .10 mm), but doing this disables the wipe tower.  I want to make sure I am sufficiently purging the support material after switching back to the PETG, but without a wipe tower this is not possible.

RickDnamps, 2022-10-31:
> ### Version
> PrusaSlicer-2.1.0-beta2+linux64-201908250731
> 
> ### Operating system type + version
> GNU/Linux Debian 9
> 
> ### 3D printer brand / version + firmware version (if known)
> Original Prusa i3 MK3s Multi Material 2S
> 
> ### Behavior
> PrusaSlicer gives a bit subversive message when trying to enable soluble supports: [image]
> 
> ```
> The Wipe Tower currently supports the non-soluble supports only
> if they are printed with the current extruder without triggering a tool change
> (both support_material_extruder and support_material_interface_extruder need to be set to 0).
> Shall I adjust those settings in order to enable the Wipe Tower?
> ```
> 
> With a bit of spelunking around in the code, it seems that wipe tower is supported, but to use it, one has to set "support_material_contact_distance" to 0 : [image]
> 
> #### Project File (.3MF) where problem occurs
> This is not about a project, but print settings present, so no project included
> 
> I propose the text to be changed to include the information about the support_material_contact_distance.

You did my day!!! Thks

kiddocteur, 2022-11-12:
Hello, is it possible to change Prusaslicer to stop restricting us with this error? 
I have the same problem as the people before: impossible to have a support material with wipe tower without the top Z contact being at 0.
Thank you

TheJoyofPrinting, 2022-11-26:
i have an issue about htis setting but i dont care about the soluble material i want my support to be a specific color and i cannot do that because this setting with the extruder for support needing to be 0. for example im using the mmu2s and have a model that is off the bed and a lot of support is needed. its wanting to use my "expensive" color i want and need for other projects and will not let me tell it to use extruder 5 which i want to put in a color i never use and dont care if its wasted on support. if i set support to any extruder besides 0 it removes the wipe tower. well my request was closed so its clear they only care about doing what they want and do not care to fix or update things the community wants...

klack, 2023-05-21:
+1 To remove this restriction.

antimix, 2023-08-28:
I can see that even in the new PrusaSlicer 2.6.1 RC this issue is still present.

I want to use PLA as support material for PET (because it does not stick well on PET and supports are easier to remove) but still if force me to treat as a "soluble" material with Contact Z=0, otherwise it removes the purge tower and then PLA and PETG are mixed without purge into the object.

Is not clear to me what would happen to the print if on the source code the check would be disabled. It would cause artifacts at layer level because the nozzle is first raised and then lowered when back to the original material ?
I have no idea.

dotCID, 2024-01-11:
As far as I understand this conflict between insoluble support using multiple extruders and also having a wipe tower comes from needing to have the layers printed by all extruders synchronised in order to have a wipe tower. Seeing as this tickbox is only available when the contact distance is 0, only soluble support mode is supported in this way.

Would it be possible to add another option there, where the support contact distance is equal to the layer height? That way the synchronisation is not disrupted, and breakaway support is once again possible.

WeaselWasher, 2024-09-11:
Another issue with this, is that if I'm using PLA for TPU support  - and set it at soluble - the interface layer is parallel lines, which I found was a little loose for the TPU to have a nice surface. I wanted to experiment and try to have some thickness to my interface layer and maybe concentric lines - to have the slicer generate this I found I had to set this above 0. (0,05 was what I wanted to try).
However, I wanted to have the wipe tower - because  - tpu.

closed 2026-09-29 by prusaslicer-bot[bot]