prusa3d/PrusaSlicer #2890: Unexpected Printer Reset - SD Card & USB Printing

resistr, 2019-09-08:
### Version
generated by PrusaSlicer 2.0.0+win64 on 2019-09-07 at 20:10:02
Version: 2.0.0+win64
Build: PrusaSlicer-2.0.0+win64-201905201652

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3 w/ MMU2 upgrade 

Printer FW: 3.7.2 
MMU FW: 1.05

### Behavior

####Actual:
Printer heats; bed levels; rehomes; but does not start the print. Printer resets after rehoming. 

####Expected: 
Printer starts printing after rehoming. 

This was a test for me to learn F360; so all project files are attached. 

### Project Files 
[Example.zip](https://github.com/prusa3d/PrusaSlicer/files/3587705/Example.zip)

Jebtrix, 2019-09-09:
The gcode is normal. This is hardware related. A better place for support is: 
* https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/

rtyr, 2021-02-03:
Not PrusaSlicer related issue. Closing.

closed 2021-02-03 by rtyr