prusa3d/PrusaSlicer #2891: Strange Travel moves

FlailAway, 2019-09-08:
### Version
2.1.0-beta3+linux64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

I don't know what that means.

### Operating system type + version
Mint 19.2, Cinnamon

### 3D printer brand / version + firmware version (if known)
Anycubic Mega, Gen-L board, Marlin 1.1.9.

See attached pics. The only thing changed between the two was reducing the external chamfer by 1-mm and that thinned out the corners. I have "Avoid crossing perimeters," checked and it is clearly not honouring that. I am printing the thicker corner ones, but I wanted to use the thinner corner version, but printing with PETG I would get a ton of stringing with those external Travel moves.

[image]

[image]

lukasmatena, 2019-09-08:
Could you please retest in PrusaSlicer2.1.0-rc? This seems to be a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2857, a preview bug from beta3 which has been already fixed.

FlailAway, 2019-09-11:
Thanks, but it is worse. In the pic, the box with all the travel moves is the one with thinner corners

The lower box corners are the same as the general wall thickness, 2mm. At the front, it now has some strings in the air. They look to be about 2mm outside of the box wall.

Both version .obj files load in as Manifold.

[image]

The config.ini, if any help, had to rename it config.log as .ini is not allowed apparently.
[config.log](https://github.com/prusa3d/PrusaSlicer/files/3601555/config.log)

FlailAway, 2019-09-11:
Here they are with "Printable" on/off for each. I am not sure if it is important, but these are loaded in with the flat surface up. So I flip them to put the flat surface on the bed.

[image]

[image]

bubnikv, 2019-09-11:
> "Printable" on/off for each.

What do you mean with that?

Some travels are possibly caused by the gap fill being performed after the perimeters.
Other travels are caused by the "avoid crossing perimeters" feature that you have enabled. You may try to disable it.

Jebtrix, 2019-09-13:
This problem is controlled/caused by seam placement. Chamfer has greater weight in seam placement candidate positions versus a fillet. I eyeballed your model and did a quick-n-dirty version of it here with 0 gap fill. Using rear seam alignment I can control the stringing travels. Somewhat related issue [#2761](https://github.com/prusa3d/PrusaSlicer/issues/2761)

[image]

 I'm not saying there isn't a problem just some insight.

Jebtrix, 2019-09-13:
Compiled latest git version and Avoid Crossing Perimeters seems to working better across all seam types. Some slight differences depending on infill pattern but looking pretty solid. Are you still using beta3?

vintagepc, 2020-01-22:
Travel planning seems better, but I'm still getting extra unneeded motion with certain shapes and "avoid crossing perimeters" turned on. Running 2.2 alpha2.
For example:

[image]

I observed it travelling down a hook, doing some infill, travelling down an ajacent hook and back out again before continuing on to the next one where it actually extruded again. 

Is this by design? (i.e does the travel planner try to follow just inside a perimeter rather than shortest path to next extrusion?)

I can capture a video of the print tomorrow; I need to run off a few more copies of the exact same gcode.

[Cable_ladder.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/4094989/Cable_ladder.gcode.zip)

bubnikv, 2020-01-22:
"avoid crossing perimeters" feature was not touched in this release. It is
ripe for a complete rewrite.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

No model has been provided so we cannot retest ourselves, however we hope that the reworked algorithm will do better.

closed 2020-12-11 by bubnikv

rkarlsba, 2021-05-19:
I'm seeing this on 2.3.1 as well as I've had before. Also using PETG, so without "avoid crossing perimiters", it becomes a mess. Watching the print now, it doesn't cross perimiters except between two corners, leaving a long string between them. This is the model https://karlsbakk.net/tmp/Planter_liner.obj