prusa3d/PrusaSlicer #2892: Height ranges feature not allowing to change Layer height settin

thomasdunn, 2019-09-09:
### Version
PrusaSlicer-2.1.0-rc+win64-201909041321

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Printrbot Metal Plus

### Behavior
* Import 3D Benchy
* Set layer height to 0.2 mm
* Right click model in Plater, Height range Modifier
* Start at height defaults to 0, Stop at height defaults to 2, Layer height defaults to 0.2
* Change Layer height in the height range UI to 0.3
* Click out of the field or tab out of the field

* _Expected Results_
  * The height range should stay at 0.3

* _Actual Results_
  * The layer height reverts to 0.2, the overall layer height for the model

Before tabbing out
[image]

After tabbing out
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[3DBenchy-height-range-not-accepting-layer-height-change_prusaslicer-2.1.0-rc.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3590993/3DBenchy-height-range-not-accepting-layer-height-change_prusaslicer-2.1.0-rc.3mf.zip)

lukasmatena, 2019-09-09:
You are right, this is a bug. There is a min/max layer height in Printer Settings->Extruder 1, which limit what layer height can be set.

[image]

You have set the max to 0, which should be taken as 75 % of nozzle diameter. This is not correctly interpreted and layer height is actually validated against zero mm, which effectively disables the layer height editing completely.

It should be an easy fix in GUI_ObjectList.cpp, static function get_max_layer_height.

thomasdunn, 2019-09-09:
OK, got it.  And thanks for the explanation.

I can confirm if I set Max Layer height limit to something other than 0, say 0.4mm, it behaves as I was expecting.

bubnikv, 2019-09-09:
> [image:
> 0-rc-screen1]
> 0-rc-screen2]

There is a maximum layer height limit at the printer->extruder settings.
Maybe you entered an excessive value not enabled by the printer profile?

lukasmatena, 2019-09-10:
This should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/4fc20090c86fc1ed11f40a8bc6a7d897012bb7bf on the *dev* branch. The fix will be part of PrusaSlicer2.2.0 or maybe a patch release of 2.1.0 if there is one.

bubnikv, 2019-12-17:
This will be fixed in PrusaSlicer 2.2.0. Closing.

closed 2019-12-17 by bubnikv

thierryzoller, 2020-10-24:
I have the same Problem in the latest build. Even changing MAX to non zero doesn't help.

lukasmatena, 2020-10-24:
@thierryzoller Thanks, but I could not reproduce the original problem in 2.3.0-alpha2. Can you please be more explicit about what "the latest build" is, what exactly is the problem and ideally post a 3MF illustrating it?