prusa3d/PrusaSlicer #2893: Random crashes

Gwedo, 2019-09-09:
### Version
PrusaSlicer 2.0.0

### Operating system type + version
Windows 10 + latest version

###Problem
Lately PrusaSlicer keeps crashing at random intervals, somtimes when launching the application, sometimes wehn slicing, sometimes when importing/exporting.

Is there anything I can do about this?

Its not on my pc side because I got I7-7700HQ, Nvidia GTX1050, 8GB ram.

With kind regards,
Guido

Jebtrix, 2019-09-09:
Your PC specs may be fine but that doesn't mean it can't be PC hardware related. You would have to attach a project that fails for you before anybody can even begin to narrow down a cause. You could also check event viewer in Windows and paste some details on the crash event.

Jebtrix, 2019-09-09:
Also try the latest release to see if behavior changes:

https://github.com/prusa3d/PrusaSlicer/releases

rtyr, 2021-02-03:
No response. Closing.

closed 2021-02-03 by rtyr