prusa3d/PrusaSlicer #2894: Vynechani linek vrchni vrstvy

AlesBolha, 2019-09-09:
### Version
2.0.0 64bit
release není vide

### Operating system type + version
W10 prof 1903

### 3D printer brand / version + firmware version (if known)

### Behavior
Po slice a prohlizeni vrstev se vse tvari ok, ale při tisku a zpetne reverzi vytvořeného kodu jsou na horni ploche vrstve sramy. Tento problem se mi prihodil uz několikrát.

[DATA.zip](https://github.com/prusa3d/PrusaSlicer/files/3591787/DATA.zip)

#### Project File (.3MF) where problem occurs
Attached

bubnikv, 2019-09-10:
Něco jako tohle?

https://github.com/prusa3d/PrusaSlicer/issues/2556#issuecomment-508555806

AlesBolha, 2019-09-10:
Dobrý den,
děkuji za zprávu vypadá to velmi podobně, včera jsem o Vás z rovna mluvil
s Petrem Zahradníkem, v tom zabaleném souboru je foto jak to vypadá u mě.

S pozdravem

Aleš Bolha | jednatel společnosti, project manager

mobil: 777 555 491 (dříve 608 314 220)

infoexe | monitoring
kontaktní adresa: K Červenému dvoru 3269/25a, 130 00 Praha 3
inforeality | infojustice | infokatastr | infodrazby | infoservis | infosms | infocar

rtyr, 2020-11-18:
Do PrusaSliceru 2.3.0-alpha byl implementován monotonic infill, který by tento problém měl řešit. Zavírám issue.

[image]

closed 2020-11-18 by rtyr