prusa3d/PrusaSlicer #2895: Input validation causes hangs for position textboxes for sub-parts

ianmcmahon, 2019-09-09:
### Version
2.1.0-beta2+

### Operating system type + version
OSX 10.13.6

### 3D printer brand / version + firmware version (if known)
RailCore 300ZL Duet

### Behavior
When working with a multi-part object, with one inner part selected, directly editing position/rotation via the textboxes on the right, non-numeric inputs cause PrusaSlicer to lock up unrecoverably.

For instance, if I accidentally put "--40" in a text box and defocus it, it will hang. 

If I put "--40" in a position text box of a single part, I get a validation error dialog, as expected.  However, if I have a multicolor object, split to parts, have an inner part selected, and put a bad number in a position text box, there's no error dialog, it instantly hangs.

#### Project File (.3MF) where problem occurs
[suit.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3592265/suit.3mf.zip)

I checked the changelog for beta3 before opening this issue, and didn't see anything related to this, so I did not try beta3.  Apologies if this has been fixed already.

Jebtrix, 2019-09-10:
I couldn't reproduce this with latest git build, on Linux at least. You may want to try using the latest 2.1.0rc release: https://github.com/prusa3d/PrusaSlicer/releases

ianmcmahon, 2019-09-10:
confirmed, I wasn't able to reproduce it on 2.1.0rc either.  However, when I upgraded, all my configs seem to have been lost to the ether :(

closed 2019-09-10 by ianmcmahon

Jebtrix, 2019-09-10:
No the profile folder is now named PrusaSlicer instead of PrusaSlicer-beta. Your profiles are still there you just have to copy the /printer, /filament, and /print folders over. Not sure where they are on OSX but on Linux its in hidden folder @` /home/username/.PrusaSlicer-beta`

ianmcmahon, 2019-09-10:
Ah yes thanks, it's ~/Library/Application Support/PrusaSlicer-beta