prusa3d/PrusaSlicer #2896: Feature Request: Print Time Statistics

dot-bob, 2019-09-09:
It would very helpful to have the print time and or estimation broken down so you could see where all the time is going and how to best optimize your prints/settings.

For instance something like:
Perimeters: 01:20:54
Top Layers: 03:56:54
Infill: 01:00:30
Retraction: 00:30:30
Raft: 00:30:20
Supports 01:33.20
Travel 00:30:00
Wipe Tower: 01:30:00
   - Color 1: 00:30:00
   - Color 2: 00:20:00
   - Color 3: 00:10:00
Etc....

Total            XXX:XX:XX

-Bob

mikolaszuza, 2019-09-11:
Cura does this, shows % of where the time is spent e.g. 23% infill, 40% supports. Really really helpful to find out what is the bottleneck time-wise.

koisdigala, 2021-01-09:
I think this has been completed in 2.3.0rc3 (earlier than that actually, but that's what I'm running now).

Not sure about the MMU info though, since I don't have one.

bubnikv, 2021-01-11:
Yes, we have it in PrusaSlicer 2.3.0.

closed 2021-01-11 by bubnikv