prusa3d/PrusaSlicer #2897: Can't arrange some objects

rongith, 2019-09-09:
### Version
PrusaSlicer 2.1.0-rc
Prusa3D Slic3r 1.41.3

### Operating system type + version
Windows 10 x64 1903

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
I can't arrange two objects. I don't know why.
I have to place them manually.
If I do arrange, they go to the center.

#### Project File (.3MF) where problem occurs
[Tunnel-r04d - I3D-1-T.zip](https://github.com/prusa3d/PrusaSlicer/files/3592611/Tunnel-r04d.-.I3D-1-T.zip)

Jebtrix, 2019-09-10:
Yeah or increase bed size to 210.19mm or higher

tamasmeszaros, 2019-09-10:
Hello and thanks for the report,
There are two problems here: 
1. Objects which do not fit on the bed should be ignored completely. This was broken as it moved them to the center and will be fixed soon.
2. The bed boundary for arrange is a bit smaller from what the UI shows as "out of bed". This is a safety offset addressing the inaccuracies of optimization in the arrangement. Unfortunately, I don't have to cycles to make this feature precise enough to allow the full bed size. The offset is very tiny, but must be there (for now at least).

As of now, arranging does not rotate any objects that is why the objects eventually end up unarranged.

rongith, 2019-09-10:
@tamasmeszaros Thank you for the explanation.
I understand your need for an offset that I wasn't aware of.
I don't need a feature to rotate automatically. My joined file was just to highlight a possible issue.

> 1. Objects which do not fit on the bed should be ignored completely. This was broken as it moved them to the center and will be fixed soon.
Just to be clear, when you talk about objects that doesn't fit on the bed, do you refer specifically to objects that doesn't fit by taking into account the offset for arrangement?

commit 5ed7d0aa (2020-03-16): Fix arrangement with items larger or equal to bed dimesions.

fixes #2897 
fixes #1186
closed 2020-03-16 by tamasmeszaros