prusa3d/PrusaSlicer #2898: Plater panel combobox long text truncation issue

Jebtrix, 2019-09-10:
### Version
version_2.1.0-rc-23-gf747b9756
wxWidgets v3.0.4
wxWidgets v3.1.2

### Operating system type + version
Arch Linux
* -DSLIC3R_STATIC=0 
* -DSLIC3R_FHS=1 
* -DSLIC3R_WX_STABLE=0 | 1
* -DSLIC3R_GTK=3 
* -DSLIC3R_PCH=0 

### Behavior
Another pesky combobox issue. The combobox will resize out of the panel. Mainly this will present itself opening a saved project and then modifying a profile with how verbose the text gets. On Linux with gtk we don't have the functionality of changing filament colors from the combobox so losing the left most side is mainly cosmetic. On Windows though the text gets truncated on the right and combobox size stays static in comparison.

[image]

bubnikv, 2019-09-10:
That's the beauty of multi-platform programming. Maybe we should fix this
issue in wxWidgets, not in slicer.

Jebtrix, 2019-09-10:
I don't envy you on the cross-platform challenge with multiple moving targets. So how is the QT version coming along? Just kidding.. just kidding. Apparently that problem is another GTK missing feature:

GtkComboBoxText needs the “ellipsize” property:
[https://gitlab.gnome.org/GNOME/gtk/issues/716](https://gitlab.gnome.org/GNOME/gtk/issues/716)

Jebtrix, 2019-09-15:
Noticed this today. Not quite the same problem but result looks similar.

[image]
[image]

bubnikv, 2021-01-08:
Implemented with 80ab47f32eae8a31849a606bf63077c5cc6248a1 for the plater preset combo boxes.
We have to review whether we still have the issue at the Fill Pattern field and similar.

bubnikv, 2021-01-08:
> We have to review whether we still have the issue at the Fill Pattern field and similar.
@YuSanka believes this has been resloved by widening these fields.
Closing.

closed 2021-01-08 by bubnikv