prusa3d/PrusaSlicer #2900: Rounding of sharp edges

espr14, 2019-09-10:
v2.1.0 rc win64

Sharp edges:
- are hard to print
- reduce printing speed
- cause higher wear of printer
- cause vibrations to surroundings
- are not necessary in a lot of models

Solution: automatic rounding of edges. For simplicity rounding in XY plane is enough so it can be processed in layers independently.

Options for user:
- radius of rounding
- min/max angle to process

These options can be enabled/disabled in modifiers. Sharp corner will be replaced by arc. If G code does not allow arcs it would be replaced by lines approximating the arc. Angle between lines would be max.angle defined by user.

If the shorter line in original model is too short, the radius would be automatically reduced to fit the arc.

This approach could be used for perimeters, sparse infill, wipe tower and supports (independently).

thinkyhead, 2019-10-15:
I have a proof of concept in the works as a G-code post-professor. It is very simple.

- Scan through the points of the complete generated layer….
- Keep track of the angular and distance change of each point as we scan….
- When the current angle+distance change matches the previous one, increment a counter….
- Above a certain threshold it must be an arc. (Too much angular change? it’s a hexagon.)
- Get the ends of the arc, the perpendicular bisector, the arc depth, the radius, direction, etc., and replace All the G0/G1 points with a G2/G3 arc.

And then -you know- a tiny leap to put real arcs into the GUI for the layer slices.
And then… G5 Bézier curves!

espr14, 2019-10-15:
Nice. Somebody made this already:
[https://github.com/DeepSOIC/SplineTravel](https://github.com/DeepSOIC/SplineTravel)

bubnikv, 2021-06-03:
Very much a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4352

awabom, 2022-12-21:
> Very much a duplicate of #4352

Using g-code arcs instead of approximating an arc with straight lines is not the same as rounding sharp corners.

closed 2026-09-29 by prusaslicer-bot[bot]