prusa3d/PrusaSlicer #2901: Ooze Prevention on dual extruders (regression)

KyleMaas, 2019-09-10:
### Version
2.1.0-rc (precompiled binary)
Problem does not exist in Slic3rPE 1.41.0 (previous version I used)

### Operating system type + version
Gentoo Linux

### 3D printer brand / version + firmware version (if known)
Heavily customized MakerFarm i3v, Marlin 1.1.0-rc6, dual heads with dual geared non-Bowden extruders and different-sized nozzles on each.

### Behavior
* _Describe the problem_
When using Ooze Prevention on a dual-material part on my dual-head machine, it works correctly until the first tool change.  When the tool change happens, the second head's temperature was applied to the first head instead.  This was my first time trying 2.1.0-rc, so I dropped back down to Slic3rPE 1.41.0, and it works correctly.

* _Steps needed to reproduce the problem_
Add part, add second part, configure one part for one extruder and one part for the other extruder, enable Ooze Prevention with a temperature differential large enough to see the change, slice, and print.

* _Expected Results_
Expected results would be like it behaved in older versions - when a nozzle is enabled, its temperature is brought up, and when disabled, its temperature is brought down by the Ooze Prevention amount.

* _Actual Results_
Upon first tool change, new set temperature for second nozzle was applied to the first nozzle instead.

* _Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Can't upload the model, as it is being printed for someone else is I am not allowed to make it public.

Jebtrix, 2019-09-10:
It works fine under a default Prusa printer profile tweaked to act like dual extruder. Because of some custom g-code handling changes and the fact you are using a custom printer profile people will need a .3MF project file to review properly. The model can just be two generic cubes unless the problem only shows on this particular model.

KyleMaas, 2019-09-11:
@Jebtrix  @lukasmatena  Probably going to take me a few days - currently need to keep the printer printing these parts (this is my only dual nozzle printer), but after that, I'll see if I can put together a synthetic test case and get back to you.

lukasmatena, 2019-10-01:
@KyleMaas Did you find a way to reproduce the bug in PrusaSlicer 2.1.0-final? Please, provide a 3MF if you did. Unless you can reproduce it yourself, the issue should probably be closed. Thanks.

KyleMaas, 2019-10-27:
Finally got some time to do more detailed debugging of this, and yes, I was able to duplicate it in 2.1.0 final.  I've attached files which I am able to duplicate it with:

[Ooze Prevention Debugging.zip](https://github.com/prusa3d/PrusaSlicer/files/3776008/Ooze.Prevention.Debugging.zip)

This .zip contains the following:
* Two test cube parts
* A .3mf of them exported from PrusaSlicer
* A config export (not a bundle, since I have tons of presets and tried to keep this test case simple)
* Exported G-Code

If you look at the exported G-Code, you can see that the M109 commands from Ooze Prevention are missing T parameters to set which nozzle they apply to.  For example:

M109 S180 ; set temperature and wait for it to be reached

Should have been

M109 S180 T0 ; set temperature and wait for it to be reached

The initial M109 commands at the start of the file do contain T parameters.  It appears to be just the Ooze Prevention ones which are missing the parameter.

KyleMaas, 2019-10-27:
I should note, too, that before posting I did some manual editing of comments in the G-Code and config export due to "After layer change G-Code" having the name of a friend of mine who I figured might not want his name in there.  But all functional G-Code is the same, so using that config export should get you the same output if you need to confirm it.

KyleMaas, 2019-11-01:
@lukasmatena Realized I didn't ping you before in my response.  Hadn't heard anything, so I figured I'd better do that in case you hadn't seen it.

lukasmatena, 2019-11-01:
@KyleMaas  Thanks. I did see it, just didn't have time to look into it properly. It is in the queue :-)

KyleMaas, 2019-11-01:
@lukasmatena Awesome.  Thanks!

KyleMaas, 2019-11-16:
@lukasmatena I have a suspicion this change might be impacting it:

https://github.com/prusa3d/PrusaSlicer/commit/9d027a558e105aca8e664c3d0f1811197b833406#diff-1cc77d57a9f4dfe9483d8ff81c0e7a7f

Reason being that, as I see it, if you have multiple extruders, then no M109 should ever be generated for them without a T parameter.  Not knowing much about the codebase itself, I don't know if that variable's not getting initialized properly somewhere, but that's one of very few changes I've found to code anywhere near temperature setting generation that has been made somewhat recently.

KyleMaas, 2020-01-19:
@lukasmatena Found and fixed.  Ooze Prevention code wasn't passing an extruder parameter to the writer, so it defaulted to -1 and caused the writer to not include a T parameter.

https://github.com/prusa3d/PrusaSlicer/pull/3548

KyleMaas, 2020-01-21:
Fixed with merge of pull request #3548.

closed 2020-01-21 by KyleMaas