prusa3d/PrusaSlicer #2902: Wipe tower for dual nozzles of different sizes

KyleMaas, 2019-09-10:
### Version
2.1.0-rc

### Operating system type + version
Gentoo Linux

### 3D printer brand / version + firmware version (if known)
Heavily customized MakerFarm i3v, Marlin 1.1.0-rc6, dual heads with dual geared non-Bowden extruders and different-sized nozzles on each.

### Behavior
* _Describe the problem_
I would love to use the wipe tower feature, but I generally run this printer with two different-sized nozzles (for single-nozzle prints, I can choose an extruder at slicing time and don't have to physically change nozzles, and many of my dual-material prints only need small detail in one of the materials, so running with two different-sized nozzles speeds up print time considerably).  Prusa Slicer 2.1.0-rc does not let you use a wipe tower with this configuration.

* _Steps needed to reproduce the problem_
Configure a printer profile with two nozzles of two different sizes.  Attempt to enable the Wipe Tower feature.

* _Expected Results_
Wipe Tower could be used.

* _Actual Results_
Error message "Ooze prevention is currently not supported with the wipe tower enabled." is displayed.

* _Is this a new feature request?_
I suppose probably.

#### Project File (.3MF) where problem occurs
Does not require an actual model, just configuring the printer and print profiles so they're in conflict.

MangyDogUK, 2019-10-05:
I would actually call this a bug rather than a feature request... Not being able to use a wipe tower when you have 2 different nozzle sizes is plane odd and unexpected. As having a different nozzle size for printing and supports is completely normal.

So yeah i would say Bug even if its a bug by omission... IE they havent written the function for it yet...

Jebtrix, 2019-10-05:
I think the issue is layer height. If you had a 1mm on one extruder and a 0.15mm on the other its gonna be much more than trivial to keep wipe tower height matched efficiently.

MangyDogUK, 2019-10-06:
In my case 0.3 and 0.4 on a 0.15 layer height....  Shouldn't be an issue.

KyleMaas, 2019-10-27:
The print layers already have to have their layer heights matched and less than the smallest nozzle's diameter for them to be able to work correctly, so I would think the layer height wouldn't be an issue for the wipe tower since you'd already be at the correct Z position.  Maybe that's simplifying things a little too much, but I've run with mismatched nozzle sizes on this machine for years without issue.

benjjyman, 2020-03-08:
This what I've been looking for dual nozzles different sizes one for infill, one for fine detail.
Someone already beaten me to it :-1: 
Print up to 3 times faster combining different hotend sizes - BCN3D Sigma R17
https://www.youtube.com/watch?v=5thn2Jpj8u8

I dont know if their profile can be made to work with other printers ???

AmyTheCute, 2021-07-09:
is this any news on this yet? my tool 0 is 0.35 and my tool 1 is 0.4mm. i don't get why i shouldn't be able to use a wipe tower but yet print in dual extrusion with the same setup!. 
same for some materials like soluble support. why does it not work? there's literally nothin to change. maybe printing the towers in two section for the latter would be beneficial but otherwise

Dakkaron, 2021-10-11:
I could really use that right now... I hope, this will be implemented some time.

carlosjln, 2021-12-07:
I'm new to dual extruder printing and I've set up my machine with 0.4 and 0.25mm nozzles precisely because I want some top layer details to be on point, but the slicer (screaming) alert was that layer height was smaller than nozzle diameter... when the smaller nozzle was to be used on the top part of the object

This should not be a problem when printing with two nozzles of different diameter and these parts are at different layer heights, meaning only one nozzle is used at any given layer height.

For the scenario where both nozzles will be used at the same layer height, then the slicer should figure out which nozzle is the smaller and use that as first nozzle on the wipe tower to print as many layers as needed to approximate 1 layer height specified for the wider nozzle.

Or simply specify that in order to use different size nozzles you can't use wipe tower nor shield, but instead a purge location with brush is necessary.

bubnikv, 2021-12-08:
@carlosjln I am wondering why after giving statements as such https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-634195468 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-634950152 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-634950152 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-642932762 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-642932762 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-667156040 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-711418420 https://github.com/prusa3d/PrusaSlicer/issues/1513#issuecomment-711421107 you are still using our product, the product developed by stubborn idiots as you stated.

Anyways, the current wipe tower implementation is biased towards the single extruder multi material setup. Indeed, there is no wipe tower in the upstream slicer and we implemented a minimum viable product (as a self proclaimed software developer you surely know what I am talking about). 

With the introduction of Prusa XL machine, we will implement a new wipe tower, which will be targeted to priming IDEX and tool changer extruders instead of single extruder multi material nozzle. Such a new wipe tower will support different nozzle diameters.

BTW you have no idea how much effort (and money, we need to feed our families) it was to enable moving the object below the print bed correctly as discussed in #1513.

carlosjln, 2021-12-08:
@bubnikv is everything ok at home buddy? 👀

This is the meaningful response:

"The current wipe tower implementation is biased towards the single extruder multi material setup. Indeed, there is no wipe tower in the upstream slicer and we implemented a minimum viable product.

With the introduction of Prusa XL machine, we will implement a new wipe tower, which will be targeted to priming IDEX and tool changer extruders instead of single extruder multi material nozzle. Such a new wipe tower will support different nozzle diameters."

The rest translates to "I'm so butt hurt that I'm going to reply to this guy with all the shit he said before, because facts hurt me". I don't need to be a "self proclaimed software developer" to understand what a "minimum viable product" is 😂, if you're still in doubt you can find my public linkedin profile ;)

Also, thanks for taking the time to reply to me :) after all this issue was reported on "Sep 10, 2019" and this is the first time you drop a comment here.

Talk about being stubborn?  that z-lift issue took you or your team almost 3 years to deal with, probably after that crap load of people commented there.

[image]

This current issue is up for anniversary too and it shows that in terms of issues and user feedback what gets resolved is whatever the company wants to do.

Finally to answer your question, why do I use PS? I have my profiles setup in Cura and PS, some models and scenarios get handled better by Cura and others by PS. You use the tools needed at any given time.

All that aside, I genuinely thank the efforts put to improve the app, software development isn't always easy nor cheap. But you can't wait 3 years to finally solve an issue.

bubnikv, 2021-12-08:
@carlosjln 
> @bubnikv is everything ok at home buddy?

To be clear. Until you learn to act respectfully, I am not your buddy.

Dakkaron, 2021-12-08:
@carlosjln I also want this implemented. But dissing a dev of a product you are using for free is not ok.

Last time I checked, this repo was open source. If you really need a feature, implement it. Or donate some money and see if you can get your feature on the roadmap that way.

But if your only contribution is personal offences, just go somewhere else. The devs here are still people and no one deserves to be attacked on that kind of personal level. You complain about @bubnikv responding directly to your attack instead of staying on topic. But you yourself started with offtopic stuff. The 23 down votes have nothing to do with the wipe tower.

carlosjln, 2021-12-08:
@Dakkaron as far as I am concerned my first comment on this post has ZERO mention to this dev, and ZERO offences or any references. So you think it is ok for someone on his working schedule and work environment (because he works for Prusa Research) to come at me out of the blue to reply to me instead of the original poster on an 2 year old issue he didn't even care about?... and you talk about being attacked on a personal level? ha! what a joke.

The 23 down votes were just a sample reminder of the reactions he got on that other issue. This one might get fixed under 3 years because prusa got money to make with the new printers coming out and they need this feature.

Visit https://github.com/prusa3d/PrusaSlicer/issues/1513 for more...

[image]

[image]

Dakkaron, 2021-12-09:
Doesn't matter. It is off topic.

Can someone with moderation privileges please clean up this thread and remove all off-topic stuff? Would be very nice.

carlosjln, 2021-12-09:
@Dakkaron I can clean up my comments, but at least have the decency to answer my first question or to admit your argument was invalid and overlooked facts. Have a good day.

Dakkaron, 2021-12-09:
@carlosjln It doesn't matter, tbh. I get why the dev doesn't like you. You are not here to improve the product, you are here to prove a point. And it doesn't even matter which point, just any point. You are disregarding everything I said, just because you want to score a point for maybe one inaccuracy. The last time I had a discussion on this level was probably in Kindergarten. "But he hit me first". I don't care. Not even a little bit. I will not answer to your posts from now on, as I will not be playing this kind of game.

Back to topic: I had a look over the code, and I can't really see a hard reason to have this limitation about the nozzle sizes. Just removing the check should give a working wipe tower. It probably will not look very good, since it will print the tower as if all the nozzles had the same size. But on the other hand, that doesn't really matter.

I will try to compile it without the check and see what the result is. I guess, it might fail if you combine a 0.1mm nozzle with a 1mm nozzle, but I am quite sure that it will work if you combine 0.4mm and 0.5mm or something similar.

carlosjln, 2021-12-09:
@Dakkaron I'm not here to improve the product? this was my first comment... learn to read comprehensively :)

[image]

carlosjln, 2021-12-09:
Probably the first and easier scenario to implement this is when one is trying to use different nozzle sizes and the smaller diameter one is used on top/last layers in order to print small fonts text or details

[image]

bubnikv, 2021-12-10:
@carlosjln 
> @bubnikv indeed, we're not buddies (that's an expression) and my respect is reserved for people who deserve it and you lost the courtesy respect given to unknown people after reading your responses to other users on that old post.

This is my last warning before you will become reported and blocked from PrusaResearch repositories.

[image]

bubnikv, 2022-01-21:
> @KyleMaas 
> many of my dual-material prints only need small detail in one of the materials, so running with two different-sized nozzles speeds up print time considerably).

Would you please provide us with real world examples where combining different nozzle diameters helped you to speed up the print? We are collecting requirements, however we don't want to over-complicate it to support some not very practical corner case scenarios.

bubnikv, 2022-01-21:
@benjjyman
> This what I've been looking for dual nozzles different sizes one for infill, one for fine detail.
> Print up to 3 times faster combining different hotend sizes - BCN3D Sigma R17
> https://www.youtube.com/watch?v=5thn2Jpj8u8

Printing objects that are perpendicular extrusions of a contour is easy, this video is a kind of cheating, they don't show how it works on general objects.

Anybody tried combining narrow nozzle for perimeters with wide nozzle for infill? Is it practical?

See for example this response: https://github.com/prusa3d/PrusaSlicer/issues/5589#issuecomment-1017810977

Dakkaron, 2022-01-22:
@bubnikv I use different nozzle diameters to work with different kinds of filament. I use a 0.4mm and a 0.5mm nozzle in my setup. The 0.4mm nozzle increases detail and gives better output on files, that were specifically optimized for that nozzle thickness (actually quite common).

The 0.5mm nozzle reduces clogging and extrusion pressure for more difficult filament like wood particle filament or flexible filaments. These filaments cause much less clogging and other problems if run on 0.5mm nozzles, since the cross-sectional area of the 0.5mm nozzle is over 50% higher than on the 0.4mm nozzle.

For example, on a certain wood filament, I had frequent clogging on 0.4mm, while I haven't had a single clog on 0.5mm.

And since I don't want to swap nozzles all the time, I have a 0.4mm nozzle on one extruder, and a 0.5mm nozzle on the other. Depending on the filament I use, I use one or the other extruder. And I don't want to have to change the 0.5mm nozzle when I run a print with 2x regular PLA, only to get the wipe tower working.

KyleMaas, 2022-01-27:
> Would you please provide us with real world examples where combining different nozzle diameters helped you to speed up the print? We are collecting requirements, however we don't want to [...]

@bubnikv As one example, I have some parts which have large flat areas with small detailed lettering on them in a different material.  These are a perfect candidate for this setup, because I don't need detail at all on the large flat areas but very much need detail on the lettering.  By running an 0.6mm nozzle (roughly 0.28mm^2) for the large flat areas vs. the 0.3mm nozzle (roughly 0.07mm^2) I use for lettering, I can lay down the background at a much faster rate than if I had to use 0.3mm for both.  Some of these parts nearly fill the print bed.  If I had to print the entire part at 0.3mm, it would take forever and I would get no extra detail out of it because of the flat surfaces and simple shapes of the background.  And this has been a setup that I've found myself printing on this printer over and over again.

Additionally, mixing sizes is sometimes necessary for combining materials, like when I print TPU and PLA or PETG in the same print.  I can print the hard plastics just fine with the 0.3mm nozzle and get excellent detail out of them but can't print TPU through that nozzle due to the backpressure finding fun new ways to kink up the filament in the feed path.  So I use the 0.6mm for the TPU and the 0.3mm for the hard plastics.

There has been at least one instance in recent memory which combined a carbon fiber filament with a glow-in-the-dark filament.  I no longer remember the exact material but it was probably PETG.  The carbon fiber was run through the 0.6mm with the glow-in-the-dark running through the 0.3mm (because it can, and I could get better detail).  The print worked out great but wouldn't have been possible with matched nozzles without sacrificing the detail on the glow-in-the-dark part.

But besides that, there are some single-nozzle prints as well for which this setup is useful.  Some prints need more detail and for some the speed of the larger nozzle is better.  The mismatched nozzle setup works great for this - if the job needs more detail, run it on extruder 1 (0.3mm).  If it's just a quick prototype or a huge part, run it on extruder 2 (0.6mm) where I can also use thicker layer heights due to the nozzle diameter being larger.  Having both nozzle sizes on the printer means I haven't had to swap nozzles on that printer in several years (they're hardened tool steel nozzles, so the wear isn't bad).

However, using this setup also means that even prints with two colors of the same material either have to be printed on the same mismatched nozzle setup or I have to swap nozzles.  I spent enough time perfectly calibrating these nozzles that I don't want to swap them.  Being able to use the wipe tower function would be a great help (especially for multi-color prints where I care about looks rather than multi-material where I only care for the function), but the tradeoff of having to swap nozzles isn't worth it for me.  The utility of mismatched nozzles is too great and it's generally not worth swapping them.

Hopefully that helps explain things a little better.  If you have any other questions or need me to expand on these, please let me know.

n8bot, 2022-01-30:
@bubnikv I use a tall draft shield for my wipe/purge tower on my IDEX printer. It works perfectly (as long as adhesion works -- addressed by my [PR Extra Skirt Base Loops](https://github.com/prusa3d/PrusaSlicer/pull/7305).

In experiments with the slicer, the tall draft shield always seems faster than a wipe tower. This is because the purge amount required is minimal with a discrete tool per material. When using multiple materials through a single tool, I concede, the wipe tower might be faster/beneficial. I have not tested that.

What I mean to say is, I think you already have the perfect "wipe tower" for IDEX/Toolchanger printers... the skirt/draft shield.

Try some experiments with the draft shield before you put too much development effort into a new wipe tower or modifying the current one. You may find that you have everything you need already.

Feel free to reach out at any time if you would like. I am excited to see what is to come for the Prusa XL and the improvements to the slicer that follow. I would be willing to volunteer development grunt work: implementing features, refactoring things at your direction. I am not an extremely capable coder, but I try to be careful and deliberate.

KyleMaas, 2022-01-30:
> What I mean to say is, I think you already have the perfect "wipe tower" for IDEX/Toolchanger printers... the skirt/draft shield.

@n8bot This is what I currently use, but in my experience it has its drawbacks.  For one, for particularly ooze-prone materials on a small nozzle, it often times does not extrude enough in one loop to fully prime the nozzle.  But the other more serious problem is that because of the priming action, the draft shield ends up being printed discontinuously which, for large parts, can result in the draft shield delaminating and curling up until it impacts the print head upon successive layers, which then foul the nozzles with melted plastic of the different head's material.  Or sometimes it will just mechanically catch on the head to the point where it starts skipping steps.  It does sort of work and is definitely better than nothing but not a perfect solution for me.

Your multi-loop skirt feature would help considerably with this problem and I think it's a great idea for even single-material parts which are very tall.  But I don't see it being able to fully eliminate the problems of successive purges not adhering properly due to randomness in starting points.  I'd love to be proven wrong and that it would work just as well, but I suspect proper wipe tower support for mismatched nozzle sizes would work better for at least some of the materials I print with which are particularly ooze-prone.

n8bot, 2022-01-30:
You are entirely correct. I was thinking to myself that I should add on to my previous comment.

To make the draft shield "work perfectly" as I said, there is a pre-purge into a "bucket" that is not over the print bed. Then, the nozzle is wiped on a silicone brush, THEN the draft shield is printed. This allows every single part of the draft shield to print correctly with minimal gaps. 

The reason this all must be done is to perfectly replicate the behaviour of normal printing. I.e., the printhead is extruding at a reasonable speed onto an existing toolpath, retracts, moves to the print, and continues. The pressure in the nozzle as closely as possible resembles any other print move.

Those initial moves, even after a purge and wipe on a brush, are never perfect. The pressure must be built up in the nozzle in a print-like way.

You can ensure that each nozzle is primed the same amount by requiring a minimum number of loops (which will be split between the tools evenly) and ensure that there is a minimum amount of material extruded for the skirt, which will also be split evenly among the extruders (with one or more possibly printing MORE, but not less than the stated amount).

KyleMaas, 2022-01-30:
> Anybody tried combining narrow nozzle for perimeters with wide nozzle for infill? Is it practical?

@bubnikv Sorry - I missed answering this question in my response.  I have not tried this particular use case.  I could see how it would be useful for some, but most of the time when I'm printing with both nozzles they are either two different materials (TPU on the 0.6mm nozzle and PETG on 0.3mm, or soluble support material on 0.6mm and ABS on 0.3mm for better detail of the final part, for a couple examples) or two different colors of the same material (glow-in-the-dark PLA panel with black PLA lettering, for example).  So I can't speak to the use case of one material and color for both just for the ability to print infill faster with small high detail perimeters.  Generally when something like that would be of use to me, my extruders are powerful enough that I just crank the extrusion width on the infill to a larger diameter than the nozzle and just push a larger volume of plastic through the small nozzle.  The utility of this for me is more for multi-material.

n8bot, 2022-01-30:
> My printer has neither a purge bucket nor brush. That generally takes up space within the print envelope

Yes, luckily I designed a bit of extra room for exactly this purpose. I didn't leave enough, though! It's expensive real estate... with all the linear rails and stuff.

I will do my best to try and remember to tag you if that idea comes up again. Right now, everything seems to work decently even when there is a bit of warping -- but I forsee larger prints, with even larger flat areas on the draft shield to become much more problematic.

And, on the topic at hand of multiple nozzle widths for the purge/wipe tower: My draft shield panacea does not properly account for multiple nozzle widths! D'Oh.

So for the topic at hand, my solution is not at all useful or ideal. But in the realm of purge/wipe towers in general, perhaps we could focus attention on making the skirt account for multiple extrusion widths and nozzle diameters. Then, find a way to pre-purge and wipe in a sensible manner for those without the luxury of off-bed space to do so.

Perhaps, the purge can be done just outside of the draft shield, onto the print surface or whatever happens to be there, and then wipe against the already printed portion of the draft shield? It definitely is more risky, but it might be a solution.

carlosjln, 2022-01-30:
Also remember simple cases like what I do for my clients, keychains and dog tags where the base is one color (0.4 nozzle) and the small letters, numbers and logo details need to be printed with 0.25 to 0.3mm nozzles. I don't want to swap nozzles or filament mid print.

n8bot, 2022-01-30:
> Also remember simple cases like what I do for my clients, keychains and dog tags where the base is one color (0.4 nozzle) and the small letters, numbers and logo details need to be printed with 0.25 to 0.3mm nozzles. I don't want to swap nozzles or filament mid print.

Yes, this was mentioned [here, by KyleMaas](https://github.com/prusa3d/PrusaSlicer/issues/2902#issuecomment-1022783021).

KyleMaas, 2022-01-30:
> Yes, this was mentioned [here, by KyleMaas](https://github.com/prusa3d/PrusaSlicer/issues/2902#issuecomment-1022783021).

Never bad to have more people with similar use cases!

taxilian, 2023-11-27:
This is closely related to #11462 and is now relevant for users with the Prusa XL

m10d, 2024-07-19:
has there been anything articulated by Prusa and/or maintainers of PrusaSlicer on where this feature sits on the roadmap / possible priority to address?

Dakkaron, 2026-09-02:
The issue is still relevant.