prusa3d/PrusaSlicer #2903: Support Interface printed on air

IgnacioHR, 2019-09-11:
### Version

PrusaSlicer 2.1.0-rc+

### Operating system type + version
OS-X

### 3D printer brand / version + firmware version (if known)
MK3S + MMU2S - 3.8.0

### Behavior

As you can see on the picture, there is support material interface printed on air.

I'll fix this for my print by changing the layer in contact with the base. I just wanted to inform you about this issue

#### Project File (.3MF) where problem occurs
[image]
[Carcasa-camara.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3600295/Carcasa-camara.3mf.zip)

bubnikv, 2021-02-22:
PrusaSlicer 2.3.0 stil shows some floating supports, see the following screenshot, right bottom corner:
[image]

The master fixes it:
[image]

bubnikv, 2021-02-22:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/3985d7bb81ddd542612ec2e8704622be4e90f074

closed 2021-02-22 by bubnikv