prusa3d/PrusaSlicer #2904: Blurry icons

MadMonkey0101, 2019-09-11:
All the icons along the top bar (add, delete etc) are blurry and seem out of focus,

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-09-11:
It would help to know what exact version PrusaSlicer your referring to and what OS your running it on. An attached screenshot would be best. There is naturally some smoothing from anti-aliasing but maybe yours is excessive. Don't know without a screenhost. Do you have any forced OS wide global overrides for GPU settings, ie FXAA Antialiasing, etc..?

If you click Help > Systeminfo > Copy to Clipboard from PrusaSlicer it would provide details you can post here.

bubnikv, 2019-09-11:
Also there is now a setting in preferences, where you may scale the icons
up or down. Scaling down will blur edges of those SVG rendered icons. We
are aware that it is not ideal, but it is non-trival to solve without
providing pixel aligned renders of all icons at various resolutions.

MadMonkey0101, 2019-09-11:
It seems scaling down to 96% cleaned up the edges.. Was totally a trivial issue. Love all the developments so far though!

malcinator, 2019-09-11:
I have had the same issue on macOS 10.14.6. I found scaling by using even numbers kept the icons relatively blur free. Of course the smaller that are the more blur there is. I have mine set to 74% and the look much better than using 75%.
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]