prusa3d/PrusaSlicer #2911: Change location of config directory

Salamandar, 2019-09-13:
On Linux, software configuration should be in ~/.config :

https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html

>  $XDG_CONFIG_HOME defines the base directory relative to which user specific configuration files should be stored. If $XDG_CONFIG_HOME is either not set or empty, a default equal to $HOME/.config should be used.

Jebtrix, 2019-09-13:
I see so many programs that don't adhere to that "standard" but you are correct. It would help make show hidden in home folder not scroll off the screen. Just have to get about 20 other programs to do the same :smile:

bubnikv, 2019-09-13:
PrusaSlicer adheres to a "standard", only to an old one.

Salamandar, 2019-09-16:
@bubnikv Let's adhere to a newer and cleaner standard ?

duramson, 2020-10-19:
The Freedesktop.org XDG base directory specification have good de facto
adoption.
It has been adopted by:
- GNOME ( https://live.gnome.org/GnomeGoals/XDGConfigFolders )
- GTK+ ( https://bugzilla.gnome.org/show_bug.cgi?id=646631 )
- KDE (
http://techbase.kde.org/KDE_System_Administration/XDG_Filesystem_Hierarchy#Freedesktop.org_and_Standard_Directories
)
- QT (
http://harmattan-dev.nokia.com/docs/library/html/qt4/qsettings.html#setPath )
- XFCE ( http://docs.xfce.org/xfce/xfce4-session/advanced in Files and
Environment Variables )
- LXDE
- Razor-qt
- VLC ( https://trac.videolan.org/vlc/ticket/1267 )
- GStreamer ( https://bugzilla.gnome.org/show_bug.cgi?id=518597 )
- Chrome ( http://code.google.com/p/chromium/issues/detail?id=16976 )
- many more upstream applications
- Ubuntu ( http://brainstorm.ubuntu.com/idea/6557/ &
http://packages.ubuntu.com/fr/source/precise/libxdg-basedir )
- Debian ( http://packages.debian.org/squeeze/libxdg-basedir1 )
- Red Hat
- Fedora
- Suse
- many more distributions

I think that OpenSSH should use same locations than the vast majority of Desktop environment and applications.

There are real advantages of following this specification :
- a lot less cluttered $HOME
- Make backups a lot more safer and easier.
  Backuping your $XDG_DATA_HOME along with your files is enough 
  (or just excluding $XDG_CACHE_HOME)
- A lot easier to reset a default configuration if you want/need it
  (and without any risk to loose informations).
  Even for the software itself could choose to reset $XDG_CONFIG_HOME if needed.
- Avoid some strange bugs that happens because you had a old version of 
  some configuration file
- A lot more of flexibility and portability because no path are hardcoded.

bubnikv, 2020-10-19:
Maybe we just need to call
https://docs.wxwidgets.org/trunk/classwx_standard_paths.html#af5eb9a009b9fd81b748e26d099bb4e89
?

bubnikv, 2020-10-19:
Somewhere here before calling datadir?

void GUI_App::init_app_config()
{
	// Profiles for the alpha are stored into the PrusaSlicer-alpha directory to not mix with the current release.
	SetAppName(SLIC3R_APP_KEY);
	SetAppName(SLIC3R_APP_KEY "-alpha");
//	SetAppDisplayName(SLIC3R_APP_NAME);

	// Set the Slic3r data directory at the Slic3r XS module.
	// Unix: ~/ .Slic3r
	// Windows : "C:\Users\username\AppData\Roaming\Slic3r" or "C:\Documents and Settings\username\Application Data\Slic3r"
	// Mac : "~/Library/Application Support/Slic3r"

	if (data_dir().empty())
		set_data_dir(wxStandardPaths::Get().GetUserDataDir().ToUTF8().data());

lukasmatena, 2020-10-23:
Commit 057c8a1c1ddaaf94dc7056300f23e326ee08eaaa should change the default datadir location on Linux in accordance with the XDG specification. User should be alerted in case the new dir does not exist and the old one does.

It would be nice if someone could compile it and check that it works properly. It does on my system, but more testing would make me happier.

Closing.

closed 2020-10-23 by lukasmatena

Salamandar, 2020-10-23:
Amazing !! :D