prusa3d/PrusaSlicer #2912: [Feature Request] Display printing time per layer

rongith, 2019-09-13:
### Version
PrusaSlicer 2.1.0-rc

### Behavior
Slic3r was never per layer edit/information oriented but improves.
It would be nice to have the printing time per layer information somewhere. Like gcode.ws and others do.
A colored View could be added and fits better the PrusaSlic3r UI as it is for now. But numeric values are of better use.
I need it when printing small parts. Then I have to analyse the PrusaSlic3r gcode file with third parties. tools/soft to adjust the settings.

lukasmatena, 2019-09-13:
I believe this to be a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2754
Could you close this issue if you agree?

rongith, 2019-09-13:
Thank you. I searched but missed that issue. There are so many issues that I'm reluctant to open new one. But as I understand @rcmaniac25 would like a timer per layer.

My need is to get the printing time of a layer, not the time since the beginning. Let's say I want to print small parts made of ABS or PC. You wouldn't go under let's say 120s per layer to get good results as the heat can't dissipate easily in small parts, especially for layers closer to the bed. The minimum print speed is the parameter that prevent you to do so. Then you can multiply small parts on the plate to increase the printing time per layer. It would be easy to see after slicing to adjust the number of parts you need to have a minimum of 120s per layer of printing time.
The features could be merged with the timer for numerical values.
A 3D color rendered view of the time per layer can also be added.

A workaround to target the 120s per layer would be to send M601 - G4 S100 - M602 in lthe custom ayer_gcode to print only one part but I'm afraid of the oozing. Never tested.

lukasmatena, 2019-09-13:
> But as I understand @rcmaniac25 would like a timer per layer.

I would consider this to be the same issue, when it gets implemented, it will certainly allow to show both (individual layer time and the accummulated sum).

> New cooling feature that say the minimum time per layer and add some pause to target the minimum time per layer

There is a parameter in Filament Settings->Cooling->Cooling Thresholds that allow you to set a minimum time per layer and will slow down the print automatically so it is not shorter. It does not stop and wait, I think to prevent oozing. Does it not suit your needs?

rongith, 2019-09-13:
> There is a parameter in Filament Settings->Cooling->Cooling Thresholds that allow you to set a minimum time per layer and will slow down the print automatically so it is not shorter. It does not stop and wait, I think to prevent oozing. Does it not suit your needs?

No. Because as I said "Min print speed" which has a priority prevents you to reach the minimum time per layer.
Exemple:
You have 10mm of 1.75 filament extruded for one layer.  That makes 24mm3. For a 0.2 layer and with a 0.4 nozzle (section is ~0.08mm2,), that makes 300mm of extruding path only. If you have a 15mm/s Min print speed, the time per layer is ~20s (given mainly time is for extruding path) which is far too low compared to the 120s I would like.

gcode.ws and octoprint give easily those information.

TickTock-, 2019-09-20:
I closed my issue as it was rather similar.  Would like to add my +1 here and see a variable made available for macro use which included this layer time to give maximum flexibility.

rongith, 2019-09-20:
> see a variable made available for macro use which indicates the layer time to give maximum flexibility.
Me too to being able to do some M601 - G4 S120-[layer_time] - M602

TickTock-, 2019-09-21:
Need to ensure the layer_time estimate includes color change time for MMU prints.  My first desire for this is the result of seeing a visible line across my prints the first time a second color is introduced.  I will play with the Filament Settings->Cooling->Cooling Threshold feature @lukasmatena mentioned on my next print.  It is possible the min print speed would not be an obstacle for my case.  None-the-less, I think adding the layer_print_time scalar would be the most flexible and enable us to deal with a lot of different scenarios.

kdsohn, 2019-10-20:
I also would appreciate the a parameter layer_print_time scalar very much. I have tried all available settings, but nothing comes close to a constant layer time (i.e. 90 - 120 seconds) regarding print quality with precious material as ABS, ASA, Nylon, PCABS or PC.
To prove that, I have made a small post processor, that calculates the layer print time and inserts variable G4 wait commands. It uses a sacrificial tower (first object to printed i each layer) to wipe off the ooze and prime the nozzle. It works like charme, but it would be much more comfortable, to have it included in the slicer.
You are welcome to test it yourself, a description is included. (https://github.com/kdsohn)
Part with waste tower and moves to waiting position:
[image]

Waste tower with collected ooze:
[image]

khumarahn, 2020-04-06:
I would also very much like a variable " layer_print_time". Printing delicate parts, I can move the nozzle away from the bed, wait for 60 seconds say, then extrude a bit, wipe it on a brush and continue printing. But I only need this on short layers... The layer time estimation is already done in the slicer, and can be inferred from the fan speed, but it would be great to have it explicitly available.

khumarahn, 2020-04-07:
Hi, I made a fork of prusa slicer with a variable 'past_layer_print_time' available in layer change scripts, see my commit at https://github.com/prusa3d/PrusaSlicer/commit/8818bb7b59e47f7a1c82b760331ea14d682d0210

This seems rather straightforward.

supermerill, 2020-06-24:
Hi I've merged your commits about this feature in my fork. 
I also added a "chronology" to see when a layer will be printed.

rongith, 2020-06-28:
I wasn't aware issues for a repository is disabled by default. I just enabled it.

Drachenkaetzchen, 2021-02-23:
I I have added support to display the estimated layer printing time. The pull request can be found in #6115 .

[image]

rongith, 2023-04-04:
Feature added. Well done Prusa!

closed 2023-04-04 by rongith

AssiveMass, 2023-04-26:
@rongith What is the variable for layer time? I couldn't find a mention of it in the [2.6alpha6 release notes](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6) or the [Macro Language page](https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-Macro-Language). Thanks!

rongith, 2023-04-26:
I went through the whole issue and figured I misunderstood my own issue. :-/ . I thought at first it was related to my old PR which display colored layer time on the part. I used to use some other strategies and forgot about it. But I would really be happy if implemented. It is a pain to print very few small parts at high temperature.

Indeed it is not supported as far I can see.

I just renamed the issue and reopened it.

flooddog, 2024-01-19:
[I can't believe anyone else has the exact same CocoaPress question... "small world but I wouldn't want to paint it!"]

I am also looking for a pause if the layer is too fast to allow the chocolate to cool. I've noticed that when a layer is very small the code sends it into a Vase Mode run where it doesn't even do a 'layer change'. When I try to print a chocolate CN Tower the very top oozes because it's a tiny tip and doesn't have time cool. 

I have added G4 pause code and that made a huge difference but I had to do it manually (search and replace gcode text at layer changes) and it doesn't take into account the layer time. I've never used a GCode IF statement so maybe that's the route?

So any help would be appreciated...

rongith, 2026-09-07:
Still a feature request.