prusa3d/PrusaSlicer #2913: Provide windows installer without drivers/firmwares/test files

Salamandar, 2019-09-13:
The current installer is 303Mbytes. That's huge.

It looks like the test files make up for most of the installer size.
Without installing them, the install dir is only 120MB and 58MB compressed. Almost 6 times less than the original exe !! And yes, even with "tools" and Pronterface selected.

I'm sure some people use those "test files", but that's certainly not the case of most people. I didn't even know those existed until today.

A thin installer should be provided, without those test files.

bubnikv, 2019-09-13:
you may just unpack the binaries zip then.

Jebtrix, 2019-09-13:
In regions where data usage is metered and not inexpensive I believe separating samples is a sensible and considerate delivery option.

Salamandar, 2019-09-16:
@bubnikv That's not really user-friendly.

Salamandar, 2020-04-23:
The installer is now 380MB, against a 38MB zip file !

rtyr, 2021-01-23:
New standalone installer has roughly 64MB. Closing.

closed 2021-01-23 by rtyr